aircrack-ng/rtl8188eus #12: Unable to change mac

BeingDatta, 2020-01-29:
Installed this driver correctly but unable to change mac

kimocoder, 2020-01-31:
Will take a look at this shortly. I'll get back.

BeingDatta, 2020-02-02:
Its solved but can you implement this in your driver
1. **sudo su**
2. **echo "options r8188eu rtm_initmac=11:22:33:aa:bb:cc" > /etc/modprobe.d/r8188eu.conf** (it will create a r8188eu.conf file inside /etc/modprobe.d folder
3. **modprobe -rv r8188eu** (to unload module)
    **modprobe -v r8188eu** (to reload it)
Dont need restart, after this just attach wifi and pass

i**fconfig wlan0 down
macchanger -r wlan0
ifconfig wlan0 up**

Its tested worked for me. Anyway Thank You for such magnificent driver.

Regards

closed 2020-02-02 by BeingDatta

kimocoder, 2020-02-02:
You know the "r8188eu" is the kernel driver, right? our module is named "8188eu", and the r8188eu should be blacklisted.