aircrack-ng/rtl8188eus #14: Black-list the Driver

Neo-vortex, 2020-02-03:
I had to change  :
"echo "blacklist r8188eu.ko" > "/etc/modprobe.d/realtek.conf"
to:
"echo "blacklist r8188eu" > "/etc/modprobe.d/realtek.conf"
to work on my machine.
my system : 
"Linux Lab 5.4.0-kali3-amd64 #1 SMP Debian 5.4.13-1kali1 (2020-01-20) x86_64 GNU/Linux"

kimocoder, 2020-02-04:
Done

closed 2020-02-04 by kimocoder

Hamstek7, 2022-05-23:
> I had to change : "echo "blacklist r8188eu.ko" > "/etc/modprobe.d/realtek.conf" to: "echo "blacklist r8188eu" > "/etc/modprobe.d/realtek.conf" to work on my machine. my system : "Linux Lab 5.4.0-kali3-amd64 #1 SMP Debian 5.4.13-1kali1 (2020-01-20) x86_64 GNU/Linux"

how to white I had to change :
"echo "blacklist r8188eu.ko" > "/etc/modprobe.d/realtek.conf"
to:
"echo "blacklist r8188eu" > "/etc/modprobe.d/realtek.conf"
to work on my machine.

Hamstek7, 2022-05-23:
please! help me to white list my WIFI adapter.? in kali linux.