aircrack-ng/rtl8188eus #17: Instruction for DKMS install

bertandzobrist, 2020-02-05:
Hi man,
could you be able to share the commands to install this driver in dkms way because I saw some dkms files inside the folder.
Regards

Si-i3ima, 2020-02-05:
chmod +x dkms-install.sh
sudo ./dkms-install.sh

Vaibhav9170, 2020-03-17:
it doesn't works by DKMS it shows error.

Jibun-no-Kage, 2022-05-09:
You can also do the following... Git the source, copy the git sources to /usr/src/.  Then do a dkms add, dkms build, then dkms install.  For example, I do the following via Anisble or scripting...

lsmod | grep r8188eu
rmmod r8188eu.ko
echo 'blacklist r8188eu' > '/etc/modprobe.d/realtek.conf'
apt-get -y install raspberrypi-kernel-headers bc build-essential libelf-dev dkms
rm -rf /home/pi/8188eu-20220319
cd /home/pi/ && git clone https://github.com/aircrack-ng/rtl8188eus.git
mv /home/pi/rtl8188eus /home/pi/8188eu-20220319
ed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' /home/pi/8188eu-20220319/Makefile

[ ! -z "$(uname -a | grep -iEw 'armv8|aarch64' )" ] && sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' /home/pi/8188eu-20220319/Makefile

[ -z "$(uname -a | grep -iEw 'armv8|aarch64' )" ] && sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' /home/pi/8188eu-20220319/Makefile

cp -rf /home/pi/8188eu-20220319/ /usr/src/rtl8188eu-5.3.9
dkms add -m rtl8188eu -v 5.3.9
dkms build -m rtl8188eu -v 5.3.9
dkms install -m rtl8188eu -v 5.3.9
rfkill
fkill unblock wlan
ip link
iw $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') info
iwlist $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') scan
iwconfig $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ')

[ -z "$(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ')"  ] && /usr/sbin/wpa_supplicant -B -D wext -i $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') -c /etc/wpa_supplicant/wpa_supplicant.conf