aircrack-ng/rtl8188eus #18: Building for NeHunter/Android

Si-i3ima, 2020-02-06:
So i build kernel headers and then i trying to build driver itself.I've met such a mistake.
`makemake ARCH=arm64 CROSS_COMPILE=../android_kernel_dark_ages/toolchains/aarch64-linux-android-4.9/bin/aarch64-linux-android- -C /lib/modules/5.4.15-arch1-1/build M=/home/i3ima/Projects/kernel/rtl8188eus  modules
make[1]: Entering directory '/usr/lib/modules/5.4.15-arch1-1/build'
make[1]: ../android_kernel_dark_ages/toolchains/aarch64-linux-android-4.9/bin/aarch64-linux-android-gcc: Command not found
Makefile:653: arch/arm64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/arm64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.4.15-arch1-1/build'
make: *** [Makefile:2301: modules] Error 2`
What i can i do?

imag1nator, 2020-03-13:
I have same error

imag1nator, 2020-03-13:
Do you have the issue?

closed 2020-05-15 by Si-i3ima