aircrack-ng/rtl8188eus #20: WN722N injections doesnt work

yoosYyd, 2020-02-06:
Monitor mode works fine.But if try some unctionality for injections, from aircrack.WiFi module lose and cant locate BSSID of target.

> CH 11 ][ Elapsed: 6 s ][ 2020-02-06 21:40                                         
                                                                                                                                   
 BSSID              PWR  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
                                                                                                                                   
 F4:F2:6D:4B:45:64  -66       16       60    0   1  135  WPA2 CCMP   PSK  TP-LINK_4564                                             
                                                                                                                                   
 BSSID              STATION            PWR   Rate    Lost    Frames  Probe                                                         
                                                                                                                                   
 F4:F2:6D:4B:45:64  CC:B8:A8:C1:FB:90    0    0 - 0e   139       60                                                                 

user@user-VirtualBox:~/Desktop$ sudo aireplay-ng --test -e TP-LINK_4564 -a F4:F2:6D:4B:45:64 wlxd037454d9936
21:40:29  Waiting for beacon frame (BSSID: F4:F2:6D:4B:45:64) on channel 6
21:40:39  No such BSSID available.
user@user-VirtualBox:~/Desktop$ sudo aireplay-ng --test -e TP-LINK_4564 -a F4:F2:6D:4B:45:64 wlxd037454d9936
21:40:43  Waiting for beacon frame (BSSID: F4:F2:6D:4B:45:64) on channel 6
21:40:53  No such BSSID available.
user@user-VirtualBox:~/Desktop$ uname -r
5.3.0-29-generic

aircrack-ng, 2020-02-06:
You're channel hopping, so the card is probably on another channel. We can also see the AP is on channel 1 but airodump-ng is on channel 11.

yoosYyd, 2020-02-07:
aircrack can't control channel hopping ? How it possible ? "airmon-ng check kill" give empty list

aircrack-ng, 2020-02-07:
You can set a fixed channel in airodump-ng  using -c. Check the documentation ([wiki](https://aircrack-ng.org/doku.php?id=airodump-ng) and manpage).

I'll assume user error and close the bug.

closed 2020-02-07 by aircrack-ng