aircrack-ng/rtl8188eus #22: request support for rtl8188ftv

ahmed-yousef, 2020-02-07:
Hello, may I request support for chipset **RTL 8188FTV** I think it is different from **8188ETV** because I applied instructions in `5.4.0-kali3-amd64` virtual box but airmon-ng not showing the card and ifconfig is not showing it either I have checked for physical and virtual box connection and both are true with no errors `lsusb` proves it.
`make` command worked fine and `make install` output the following
```
install -p -m 644 8188eu.ko  /lib/modules/5.4.0-kali3-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.4.0-kali3-amd64
```
so, any help, if I made any mistake in installation, will be appreciated.

ysyousufahmed, 2020-04-12:
i think kali 2020.1 (kernel 5.4.0) is not compatable with chipset RTL8188FTV.
I also seen lots of forum, tried lots of things but still i am not getting the solution.

anexcho, 2020-05-25:
Is it indeed not compatible? I also tried different things but no success...

Any help will be appreciated...

maxsterb, 2020-06-05:
Same with me too 😞

Zako04, 2023-05-26:
Same too