aircrack-ng/rtl8188eus #26: Cannont load driver

nietoperz809, 2020-03-10:
Hi, I'm new to Linux. I build it but can't get the 8188eu.ko to work.

sudo insmod 8188eu.ko 
... says ... 
insmod: ERROR: could not insert module 8188eu.ko: Invalid parameters

umod -a says 
Linux kalli2 5.4.0-kali4-686-pae #1 SMP Debian 5.4.19-1kali1 (2020-02-17) i686 GNU/Linux

sudo modinfo 8188eu.ko 
gives me ...

filename:       /home/paul/rtl8188eus/8188eu.ko
version:        v5.3.9_28540.20180627
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     1C9D32A266ABA443E146ABC
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
depends:        usbcore,cfg80211
retpoline:      Y
name:           8188eu
vermagic:       5.4.0-kali4-686-pae SMP mod_unload modversions 686 
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_lps_level:The default LPS level (int)
parm:           rtw_max_bss_cnt:int
parm:           rtw_usb_rxagg_mode:int
parm:           rtw_dynamic_agg_enable:int
parm:           rtw_drv_log_level:set log level when insert driver module, default log level is _DRV_INFO_ = 4 (uint)
parm:           rtw_tx_bw_mode:The max tx bw for 2.4G and 5G. format is the same as rtw_bw_mode (uint)
parm:           rtw_rx_ampdu_sz_limit_1ss:RX AMPDU size limit for 1SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_2ss:RX AMPDU size limit for 2SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_3ss:RX AMPDU size limit for 3SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_4ss:RX AMPDU size limit for 4SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rf_config:int
parm:           rtw_country_code:The default country code (in alpha2) (charp)
parm:           rtw_channel_plan:The default chplan ID when rtw_alpha2 is not specified or valid (int)
parm:           rtw_excl_chs:exclusive channel array (array of uint)
parm:           rtw_qos_opt_enable:int
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_pwrtrim_enable:int
parm:           rtw_initmac:charp
parm:           rtw_special_rf_path:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_ht_enable:int
parm:           rtw_bw_mode:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_rx_ampdu_amsdu:int
parm:           rtw_tx_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_full_ch_in_p2p_handshake:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_drv_ant_band_switch:int
parm:           rtw_single_ant_path:int
parm:           rtw_switch_usb_mode:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_check_hw_status:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL. 0:Disable, 1:enable, 2:by usb speed (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           rtw_hiq_filter:0:allow all, 1:allow special, 2:deny all (uint)
parm:           rtw_adaptivity_en:0:disable, 1:enable (uint)
parm:           rtw_adaptivity_mode:0:normal, 1:carrier sense (uint)
parm:           rtw_adaptivity_dml:0:disable, 1:enable (uint)
parm:           rtw_adaptivity_dc_backoff:DC backoff for Adaptivity (uint)
parm:           rtw_adaptivity_th_l2h_ini:th_l2h_ini for Adaptivity (int)
parm:           rtw_adaptivity_th_edcca_hl_diff:th_edcca_hl_diff for Adaptivity (int)
parm:           rtw_amplifier_type_2g:BIT3:2G ext-PA, BIT4:2G ext-LNA (uint)
parm:           rtw_amplifier_type_5g:BIT6:5G ext-PA, BIT7:5G ext-LNA (uint)
parm:           rtw_RFE_type:default init value:64 (uint)
parm:           rtw_powertracking_type:default init value:64 (uint)
parm:           rtw_GLNA_type:default init value:0 (uint)
parm:           rtw_TxBBSwing_2G:default init value:0xFF (uint)
parm:           rtw_TxBBSwing_5G:default init value:0xFF (uint)
parm:           rtw_OffEfuseMask:default open Efuse Mask value:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse value:0 (uint)
parm:           rtw_rxgain_offset_2g:default RF Gain 2G Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int

---------------------------------------------------------------------------
what am I doing wrong? What parameters are invalid?
Thanks in advance for your help.

nietoperz809, 2020-03-10:
the "file" command says:
8188eu.ko: ELF 32-bit LSB relocatable, Intel 80386, version 1 (SYSV), BuildID[sha1]=20d873fb1151d3ebc411baa4bce2c5e568b451d8, with debug_info, not stripped

Is this ok for my Linux version?
5.4.0-kali4-686-pae #1 SMP Debian 5.4.19-1kali1 (2020-02-17) i686 GNU/Linux

nietoperz809, 2020-03-10:
I did further investigation: dmesg!

[ 3962.720832] 8188eu: disagrees about version of symbol __cfg80211_alloc_event_skb
[ 3962.720841] 8188eu: Unknown symbol __cfg80211_alloc_event_skb (err -22)
[ 3962.720855] 8188eu: disagrees about version of symbol wiphy_register
[ 3962.720857] 8188eu: Unknown symbol wiphy_register (err -22)
[ 3962.720876] 8188eu: disagrees about version of symbol cfg80211_remain_on_channel_expired
[ 3962.720878] 8188eu: Unknown symbol cfg80211_remain_on_channel_expired (err -22)
[ 3962.720918] 8188eu: disagrees about version of symbol cfg80211_vendor_cmd_reply
[ 3962.720920] 8188eu: Unknown symbol cfg80211_vendor_cmd_reply (err -22)
[ 3962.720940] 8188eu: disagrees about version of symbol cfg80211_del_sta_sinfo
[ 3962.720942] 8188eu: Unknown symbol cfg80211_del_sta_sinfo (err -22)
[ 3962.720957] 8188eu: disagrees about version of symbol wiphy_unregister
[ 3962.720959] 8188eu: Unknown symbol wiphy_unregister (err -22)
[ 3962.721118] 8188eu: disagrees about version of symbol cfg80211_ch_switch_notify
[ 3962.721119] 8188eu: Unknown symbol cfg80211_ch_switch_notify (err -22)
[ 3962.721177] 8188eu: disagrees about version of symbol cfg80211_ready_on_channel
[ 3962.721179] 8188eu: Unknown symbol cfg80211_ready_on_channel (err -22)
[ 3962.721229] 8188eu: disagrees about version of symbol wiphy_free
[ 3962.721231] 8188eu: Unknown symbol wiphy_free (err -22)
[ 3962.721239] 8188eu: disagrees about version of symbol wiphy_new_nm
[ 3962.721241] 8188eu: Unknown symbol wiphy_new_nm (err -22)
[ 3962.721247] 8188eu: disagrees about version of symbol cfg80211_get_bss
[ 3962.721249] 8188eu: Unknown symbol cfg80211_get_bss (err -22)
[ 3962.721273] 8188eu: disagrees about version of symbol __cfg80211_send_event_skb
[ 3962.721275] 8188eu: Unknown symbol __cfg80211_send_event_skb (err -22)
[ 3962.721296] 8188eu: disagrees about version of symbol cfg80211_disconnected
[ 3962.721298] 8188eu: Unknown symbol cfg80211_disconnected (err -22)
[ 3962.721309] 8188eu: disagrees about version of symbol cfg80211_michael_mic_failure
[ 3962.721311] 8188eu: Unknown symbol cfg80211_michael_mic_failure (err -22)
[ 3962.721344] 8188eu: disagrees about version of symbol cfg80211_ibss_joined
[ 3962.721346] 8188eu: Unknown symbol cfg80211_ibss_joined (err -22)
[ 3962.721429] 8188eu: disagrees about version of symbol cfg80211_scan_done
[ 3962.721431] 8188eu: Unknown symbol cfg80211_scan_done (err -22)
[ 3962.721447] 8188eu: disagrees about version of symbol cfg80211_roamed
[ 3962.721449] 8188eu: Unknown symbol cfg80211_roamed (err -22)
[ 3962.721453] 8188eu: disagrees about version of symbol cfg80211_put_bss
[ 3962.721455] 8188eu: Unknown symbol cfg80211_put_bss (err -22)
[ 3962.721476] 8188eu: disagrees about version of symbol cfg80211_connect_done
[ 3962.721478] 8188eu: Unknown symbol cfg80211_connect_done (err -22)
[ 3962.721528] 8188eu: disagrees about version of symbol cfg80211_unlink_bss
[ 3962.721529] 8188eu: Unknown symbol cfg80211_unlink_bss (err -22)
[ 3962.721568] 8188eu: disagrees about version of symbol ieee80211_get_channel
[ 3962.721570] 8188eu: Unknown symbol ieee80211_get_channel (err -22)
[ 3962.721575] 8188eu: disagrees about version of symbol cfg80211_inform_bss_frame_data
[ 3962.721577] 8188eu: Unknown symbol cfg80211_inform_bss_frame_data (err -22)
[ 3962.721590] 8188eu: disagrees about version of symbol cfg80211_new_sta
[ 3962.721592] 8188eu: Unknown symbol cfg80211_new_sta (err -22)
[ 3962.721656] 8188eu: disagrees about version of symbol __cfg80211_alloc_reply_skb
[ 3962.721657] 8188eu: Unknown symbol __cfg80211_alloc_reply_skb (err -22)
[ 3962.721669] 8188eu: disagrees about version of symbol cfg80211_rx_mgmt
[ 3962.721671] 8188eu: Unknown symbol cfg80211_rx_mgmt (err -22)
[ 3962.721681] 8188eu: disagrees about version of symbol wiphy_apply_custom_regulatory
[ 3962.721683] 8188eu: Unknown symbol wiphy_apply_custom_regulatory (err -22)
[ 3962.721742] 8188eu: disagrees about version of symbol cfg80211_mgmt_tx_status
[ 3962.721745] 8188eu: Unknown symbol cfg80211_mgmt_tx_status (err -22)

-----------------------------------------------------------------------------------------------------------------
Seems to be a problem with my compiler/makefile setup?
What can I do to fix that?

nietoperz809, 2020-03-11:
I'm now trying to install the driver on a fresh Linux system.
Maybe my cfg80211.ko is fucked up because I installed a monitor driver for rtl8187 before.

ryanjoseph999, 2020-03-11:
> I'm now trying to install the driver on a fresh Linux system.
> Maybe my cfg80211.ko is fucked up because I installed a monitor driver for rtl8187 before.

did you blacklist the previous driver?

nietoperz809, 2020-03-12:
Yes I did. Meanwhile I can load the driver.  I have to manually load cfg80211.ko before. But it still didn't work. It won't enter monitor mode. Also the LED isn't blinking like in windows. The power-on routine is called but the firmware-download code not. I don't know why ....

nietoperz809, 2020-03-15:
Meanwhile I got it running. Also monitor mode works, but the device name doesn't change to wlanXmon but remains wlanX. The only problem now is packet injection which causes a CPU exception (as shown by dmesg).

hinell, 2020-03-18:
@nietoperz809 I'm experiencing the same problem with LED indicators. Would you like to create a separate issue on that?

nietoperz809, 2020-06-09:
Hi hinell,
there is a #define that is commented out. Sorry, I forgot name and .h file where it exists. The name is something with LED (all capitals). Scan all .h files for a define continuing LED and you'll find it. The identifier is pretty self-explanatory.

sanqudui8ban, 2021-05-19:
kernel config has a item that called cfg80211 . if you set this item, the problem will be solved.