aircrack-ng/rtl8188eus #31: SOLVED !!!!!!!! Nethunter RTL8188eu and probably RTL8812AU Installation Guide

BeingDatta, 2020-03-30:
**Guide to install RTL8188EU on NetHunter**

Some Important Points: Do not change your kernel name or version in any manner while preparing the module or else it will not work and you will get errors like insmod 8188.ko >> fatal error exec format failed. This happens when you use build script from kali site and on the other hand you build your modules manually.

For example say during the time of building kernel your situation was like /lib/modules/3.10.35-kali and while making the module your position is say /lib/modules/3.10.35_kali or anything else. This kind of thing will not work. So first built your kernel then just after that build your modules.

1.Toolchain Used for building this module - gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf

2.Your kernel should enabled below. check in /proc/config.gz or device_defconfig
CONFIG_MODULES (loadable module support)
CONFIG_MODULE_FORCE_LOAD (forced module loading)
CONFIG_MODULE_UNLOAD (module unloading)
CONFIG_MODULE_FORCE_UNLOAD (forced module loading)
CONFIG_CFG80211_WEXT (wireless extension compability)
CONFIG_MAC80211 (IEEE 802.11 Networking Stack)

**Step 1- Do the following: After building the kernel stay inside that directory so the first line is irrelevant.**

Cd android kernel-sourcedirectory on your pc or laptop
mkdir ../kernel-headers (make kernel-headers directory outside kernel source)
make O=../kernel-headers someone_defconfig
make O=../kernel-headers modules_prepare (prepare module)
make O=../kernel-headers modules INSTALL_MOD_PATH=../kernel-headers (make modules)
make O=../kernel-headers modules_install INSTALL_MOD_PATH=../kernel-headers(install modules)

**Step 2- Do the following:**

make headers_install CROSS_COMPILE=$path-your-cross-compiler ARCH=arm SUBARCH=arm INSTALL_HDR_PATH=../kernel-headers (install kernel headers on the same directory ie “kernel-headers”)

-Note eg: CROSS_COMPILE=
/home/user/gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-

**Step 3- Do the following:**

	cd ../
	git clone https://github.com/aircrack-ng/rtl8188eus -b v5.3.9
	cd rtl8188eus

Now enter rtl8188eus directory using your file manager and edit the “makefile” as follows: for armhf
###################### Platform Related #######################
Line number 94-95

CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y

## For ARM ToolChain use Hardware FLOATING line number 1044-1053

ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH ?= arm
CROSS_COMPILE ?=toolchain path (give urs)
KVER ?= 3.18.35-kali	(give urs version /lib/modules/)
KSRC :=fullpath not shown include it/kernel-headers/lib/modules/3.18.35-kali/build	(give urs)
MODDESTDIR := fullpath not shown include it/3.18.35-kali/kernel/drivers/net/wireless/	(give urs)
INSTALL_PREFIX := leave as it is
Endif

**Step 4- from inside of rtl8188eus directory** 

export ARCH=arm
Make

Done enjoy your 8188eu.ko inside this same directory. Transfer it to your nethunter by reflashing kernel after including it inside modules/kernel version/. 

**Regards
[Guide line to install RTL8188EU on NetHunter.pdf](https://github.com/aircrack-ng/rtl8188eus/files/4401255/Guide.line.to.install.RTL8188EU.on.NetHunter.pdf)
**

dufduf5, 2020-05-07:
>/lib/modules/uname-r or kernel version/ folder you will find two linked directory name /build and /source enter inside them individually and copy everything inside /include folder and paste them inside the /lib/modules/uname-r or kernel version/build and /lib/modules/uname-r or kernel version/source folder respectively. Do all this in a separate place and at the end place them inside your device folder for building net hunter kernel like this

>Inside device codename folder say "Flounder/modules/uname-r/here you have to paste your build and source folder/. So inside "Flounder/modules/uname-r" folder you will have kernel, build and source folder now prepare and flash the kernel you will have your kernel headers inside /system/lib/modules/uname-r/build & /system/lib/modules/uname-r/source respectively.

This implies creating the directory modules/$(uname -r)/ manually in the device_source directory, and then copy the source and build from /lib/modules/$(uname -r) so that it turns out for example ~/android_kernel_lge_hammerhead/modules/3.4.0-8.14/build & source? Directory "kernel" I can to take from ~/kernel-headers/lib/modules/3.4.0-8.14/ too? But dirctoryes build and source it's SymLink, I need to follow a symbolic link and copy them from there? This means that I will copy the directory android_kernel_lge_hammerhead in android_kernel_lge_hammerhead? 
So, after that I have to build the kernel again so that I can upgrade it, but before assembling the driver, I already built the kernel and there is already zImage-dtb in the ~ / android_kernel_lge_hammerhead / arch / arm / boot directory. I need an explanation please, you can describe it in more detail.

BeingDatta, 2020-05-07:
once you successfully build your kernel you will find /lib/modules inside your kernel output directory if you have built any modules. then inside /lib/modules u will find a directory like say 3.10.4-kali this is ur kernel or uname-r directory

trueToastedCode, 2020-05-28:
I don't have a folder like lib/modules/@version_name ...
[kernel-headers-OP7.zip](https://github.com/aircrack-ng/rtl8188eus/files/4697303/kernel-headers-OP7.zip)

kimocoder, 2020-05-28:
This driver will be deprecated. I made a new one, much better, much easier to add to Android.. mac80211 subsystem instead of the old kernel module.

https://github.com/kimocoder/realtek_wifi

trueToastedCode, 2020-05-29:
@ kimocoder c If this works, it would be amazing. I also fixed now, that i did not had a version name ;D Very stupid mistake, i did not use modules_install at the end..

kimocoder, 2020-05-29:
Yeah, that other driver works way better.. old IO/subsystem versus mac80211 can't compete really. Easier to build into Android also. Way easier, since it's a kernel upstream driver it's based upon

trueToastedCode, 2020-05-29:
I did it finally. I put the driver in /storage/emulated/0 and loaded it and i could set it into monitor mode... I don't have a folder called /system/lib/modules.. may someone help me?

kimocoder, 2020-05-29:
[image]

BeingDatta, 2020-06-05:
> This driver will be deprecated. I made a new one, much better, much easier to add to Android.. mac80211 subsystem instead of the old kernel module.
> 
> https://github.com/kimocoder/realtek_wifi

Amazing man I will try can u do the same for rtl8812au one it is you who made this too.

kimocoder, 2020-06-05:
Don't have time for that, doing many other projects at the same time :100:

closed 2020-06-05 by kimocoder

BeingDatta, 2020-06-05:
> Don't have time for that, doing many other projects at the same time :100: 

ok man thanks for this any way

BeingDatta, 2020-06-06:
> This driver will be deprecated. I made a new one, much better, much easier to add to Android.. mac80211 subsystem instead of the old kernel module.
> 
> https://github.com/kimocoder/realtek_wifi

Dont do this man I have gone through your new driver and what I can say is that this particular driver of yours you mentioned might be good for kernel 4.0 and above but not for kernel 3.xx or 3.xx.xy variants. There are lots of devices with these 3.xx version of kernels henceforth they will loose their ability to use nethunter.

Rettrjo, 2024-02-15:
[image]
I am using gcc-linaro-7.5.0-2019.12-x86_64_aarch64-linux-gnu cross compiler for Arm64 device.