aircrack-ng/rtl8188eus #33: added kernel 5.6.0 patch

darkalexpp, 2020-04-01:
patch for kernel 5.6

[patch_added_kernel_5.6.0.txt](https://github.com/aircrack-ng/rtl8188eus/files/4415493/patch_added_kernel_5.6.0.txt)

niteshagrawal, 2020-04-05:
i am using kali 2020.1. since i am new to all these stuff. it will be very helpful if you can tell me how to use this patch. and it will be very helpful if you have any video tutorial regarding this.

darkalexpp, 2020-04-05:
> i am using kali 2020.1. since i am new to all these stuff. it will be very helpful if you can tell me how to use this patch. and it will be very helpful if you have any video tutorial regarding this.

you need to clone this git, and then copy "patch_added_kernel_5.6.0.txt" into rtl8188eus folder. so open a terminal in this path and apply with:

git apply patch_added_kernel_5.6.0.txt

alfter you can compiling and install with
make && make install

runlevel5, 2020-04-06:
LGTM +1

darkalexpp, 2020-04-06:
> LGTM +1
> 
> @darkalexcpp Could you please submit a Pull Request?

yes. I already submitted a pull request

runlevel5, 2020-04-06:
The PR was merged. Please feel free to close this issue.

closed 2020-04-06 by darkalexpp