aircrack-ng/rtl8188eus #36: Help here with the monitor mode enabling

aledebessa, 2020-04-05:
$ sudo ip link set <interface> down
bach: interface: no such file or directory

MrEngman, 2020-04-05:
try:-

$ sudo ip link set wlan0 down

aledebessa, 2020-04-05:
> try:-
> 
> sudo ip link set wlan0 down

That one worked.

But when i type the:

`$ sudo iw dev <interface> set type monitor`

I get that isn't supported. Is it correct?

As I see here, my adapter is RTL8188FTV.

darkalexpp, 2020-04-06:
> $ sudo ip link set down
> bash: interface: no such file or directory

I use:
sudo iwconfig wlan0 mode monitor

Shacklebolt13, 2020-04-09:
here, put this inside a file named monitor-mode and place in any PATH directories(i placed in /sbin)
grant it exec permissions using chmod +x <filename>
then use <filename in shell to switch monitor mode on or off>
`
#!/bin/sh
#don't remove this credit line, it's literally my first shell code 0w0
echo "Made my Gagan Singh"
on(){
 sudo iw dev wlan0 set type managed;
 sudo iwconfig wlan0 mode managed;
 sudo service network-manager start;
}

off(){
 sudo ip link set wlan0 down;
 sudo service network-manager stop;
 sudo iw dev wlan0 set type monitor
 sudo iwconfig wlan0 mode monitor;
}

if (iwconfig wlan0 | grep Monitor) then
 on
else
 off
fi
`