aircrack-ng/rtl8188eus #39: SOLVED: The driver for wn722n does not work after a reboot!!

Adonai-ai, 2020-04-08:
I installed this driver on different systems, but after rebooting the driver stopped working. The problem was fundamentally resolved: I deleted all references to the rtl8188eus driver from the computer. Search command:
`find /lib/modules -name 'rtl8188*'`
After rebooting, the driver works great!!!

P.S. I don't know much English, so I use a translator))

SmillerMP, 2020-05-21:
Do you still have the problem? Sorry i dont understand so much your post

vytatuas, 2020-05-27:
@SmillerMP Hi bruh, I'm usinig TL-WN722N, but I can not change use macchanger and I can not open monitor mode. Could you help me please If you solved.

For macchanger (macchanger -r wlan0) : Network driver didn't actually change to the new MAC!!

For Monitor mode  ( iwconfig wlan0 mode monitor) : Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.
Help me please

BeingDatta, 2020-05-27:
follow aircrack-ng rtl8188eus v 5.3.9 readme apply the same way that has been mentioned there to go to monitor mode ur macchanger -r wlan0 will work then use ip link and iw dev instead of ifconfig and iwconfig

SmillerMP, 2020-05-27:
hi dude @vytatuas 
@BeingDatta have the reson 

you have to follow the instructions, if you have problems you can follow this post, (it is in Spanish but you can translate it and follow the instructions).

- make shure you are super user

ayushnanda14, 2020-08-05:
@SmillerMP thanks it worked for me