aircrack-ng/rtl8188eus #41: suggesting patch to work with kernel-5.6.3

ohquait, 2020-04-10:
[https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/rtl8188eus-5.7.6.1-ioctl_cfg8021.c-kernel5.6.patch](url)

this work for me, clean compiled

`apt-get install dkms-rtl8188eus-5.7.6.1-2pclos2020.x86_64.rpm
Reading Package Lists... Done
Building Dependency Tree... Done
Selecting dkms-rtl8188eus for 'dkms-rtl8188eus-5.7.6.1-2pclos2020.x86_64.rpm'
The following NEW packages will be installed:
   dkms-rtl8188eus (5.7.6.1-2pclos2020)
0 upgraded, 1 newly installed, 0 removed and 19 not upgraded.
Need to get 0B/1955kB of archives.
After unpacking 14.0MB of additional disk space will be used.
Committing changes...
Preparing                                ############################## [100%]
Updating / installing
  dkms-rtl8188eus-5.7.6.1-2pclos2020.x86 ############################## [100%]

Creating symlink /var/lib/dkms/rtl8188eus/5.7.6.1-2pclos2020/source ->
                 /usr/src/rtl8188eus-5.7.6.1-2pclos2020

DKMS: add Completed.

Preparing kernel 5.6.3-pclos1 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make clean.....
using /proc/config.gz
make oldconfig.....
make prepare.....

Building module:
cleaning build area....
'make' SKIP_STACK_VALIDATION=1 all...................................................
cleaning build area....
cleaning kernel tree (make clean).....

DKMS: build Completed.

rtl8188eus.ko.xz:
 - Installation
   - Installing to /lib/modules/5.6.3-pclos1/kernel/3rdparty/rtl8188eus/

depmod.....

DKMS: install Completed.
Reloading network configuration ... 
Done.`

the modestdir is just for me changed, ...

kudws, 2020-04-19:
agree

Mukundhan98, 2020-06-20:
Can you please elaborate the steps and link to the package, I'm having trouble detecting adapter on my VM.
My Kali kernel 5.6.0. I checked my adapter in Kali kernel 5.3.0, it works without any issue. But not works in 5.6.0. It is not even showing up.

sanyam8929, 2020-07-27:
can you explain a lit bit more and provide new patch for newer version for kail 5.7.0