aircrack-ng/rtl8188eus #42: module compile error for NetHunter, Nexus 5

login94964, 2020-04-11:
I'm found NetHunter kernel source for my Nexus 5:
This https://github.com/mayankmetha/Nethunter-Hammerhead-Kernel
I'm used this toolchain:
https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-eabi-4.8
First I'm build my kernel from source, compilation was successful...
next I'm did everything just like here:
https://github.com/aircrack-ng/rtl8188eus/issues/31
And always get it's:
`neither@DESKTOP-V9HVN9E:~/rtl8188eus$ make
make ARCH=arm CROSS_COMPILE=arm-eabi- -C ../kernel-headers/lib/modules/3.4.0-NetHunter-N5/build M=/home/neither/rtl8188eus  modules
make[1]: Entering directory '/home/neither/kernel-headers'
  CC [M]  /home/neither/rtl8188eus/core/rtw_cmd.o
In file included from /home/neither/nethunter-hammerhead-kernel/include/linux/module.h:9:0,
                 from /home/neither/rtl8188eus/include/basic_types.h:76,
                 from /home/neither/rtl8188eus/include/drv_types.h:25,
                 from /home/neither/rtl8188eus/core/rtw_cmd.c:17:
/home/neither/nethunter-hammerhead-kernel/include/linux/list.h:24:42: warning: 'struct list_head' declared inside parameter list [enabled by default]
error, forbidden warning: list.h:24
make[4]: *** [/home/neither/nethunter-hammerhead-kernel/scripts/Makefile.build:308: /home/neither/rtl8188eus/core/rtw_cmd.o] Error 1
make[3]: *** [/home/neither/nethunter-hammerhead-kernel/Makefile:1372: _module_/home/neither/rtl8188eus] Error 2
make[2]: *** [Makefile:130: sub-make] Error 2
make[1]: *** [Makefile:26: all] Error 2
make[1]: Leaving directory '/home/neither/kernel-headers'
make: *** [Makefile:2047: modules] Error 2
`
with other toolchain compilers everything is also
I'm ask for help. How can i solve this?

BeingDatta, 2020-04-12:
see this exactly follow the rules
https://github.com/aircrack-ng/rtl8188eus/issues/31

BeingDatta, 2020-04-12:
also u need 4.9 for this driver 4.8 will not work or will somehow give annoying errors