aircrack-ng/rtl8188eus #43: sudo rmmod r8188.ko kills my wifi & make file error

vishu003, 2020-04-15:
when i enter command 'sudo rmmod r8188eu.ko' it always kills my wifi and when i type make it shows the following error :
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-kali2-amd64/build M=/home/vishal/Desktop/rtl8188eus  modules
make[1]: *** /lib/modules/5.4.0-kali2-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2047: modules] Error 2
please help

Imlooser, 2020-05-08:
yes me too

BeingDatta, 2020-05-08:
do you have linux kernel headers inside /usr/src directory if not then install it search from google

Imlooser, 2020-05-08:
i am having them

adhamkhaled-dev, 2020-06-01:
Try using uname -r 
To know the name of the device then type
apt-get install linux-headers-4.4.0-21-generic for example..
and see if the make command will work 
Don't forget to plug in the device before rmmod r8188eu.ko

beelzebro, 2021-04-28:
My network manager.conf file is all jacked up. Do I have to manually set up my wireless connections?

69Vishal, 2021-04-29:
Bruh this method is shit, just but a new and better WIFI antenna... :/

beelzebro, 2021-05-22:
Yeaup. Did that.

ajaynethuluri, 2022-08-21:
hi