aircrack-ng/rtl8188eus #47: Kernel not supported (5.4.0-kali)

Gudlynoob, 2020-04-30:
when i type make 
i see 

> root@kali:/home/gudly/rtl8188eus# make

`make` ARCH=arm64 CROSS_COMPILE=../toolchain/toolchain64/bin/aarch64-linux-android- -C /lib/modules/5.4.0-kali2-amd64/build M=/home/gudly/rtl8188eus  modules
make[1]: *** /lib/modules/5.4.0-kali2-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2047: modules] Error `2`

>

BeingDatta, 2020-05-03:
install kernel headers 

> sudo apt install linux-headers-$(uname -r)

you /usr/src directory dosent have kernel headers

Gudlynoob, 2020-05-10:
is this code support up to kernel 5.5 ?

Gudlynoob, 2020-05-10:
> install kernel headers
> 
> > sudo apt install linux-headers-$(uname -r)
> 
> you /usr/src directory dosent have kernel headers

> now the problem is, when i type
> 
>> sudo iw dev wlan0 set type monitor
>
> it's said 
>
>> command failed: No such device (-19)

Gudlynoob, 2020-05-10:
i used tp-link WN722n V2, and kernel 5.5.0-kali2-amd64

awkimball, 2020-05-21:
wlan0 might not be the correct interface name, try running`iw list` to get a list of your devices and their interface names. Replace `wlan0` with whatever your device was renamed to

SmillerMP, 2020-05-21:
@Gudlynoob  I used the TP-WN722N v3 in the same version of kali linux! And everything is going well! 

if you use virtual machine make sure if have the drivers for the usb 2.0 and the adapter is select on the virtual machine!    

Or confirm if the adapte’s name is the correct with the comand  _iw_ 

Good luck 

Sorry for my english

BeingDatta, 2020-06-01:
check ifconfig or iw dev for interface name because it may not be wlan0 in there or do this

nano /etc/default/grub then put net.ifnames=0 beside quiet splash.

then sudo update-grub  then restart u will get all interfaces as wlan0 usb0 wwan0 etc

diablo-huma, 2022-04-29:
You have to install the corresponding image, in my case was:

`sudo apt-get install linux-image-5.16.0-kali7-amd64`
Check it here:

https://unix.stackexchange.com/questions/328655/cant-install-linux-headers-kali-linux