aircrack-ng/rtl8188eus #49: Compilation error on rtw_mlme_ext.c

dufduf5, 2020-05-02:
I used this
toolchain:
wget https://releases.linaro.org/components/toolchain/binaries/4.9-2016.02/arm-linux-gnueabihf/gcc-linaro-4.9-2016.02-x86_64_arm-linux-gnueabihf.tar.xz
kernel source:
git clone https://github.com/Re4son/android_kernel_lge_hammerhead -b nethunter-7.1
p.s I uploaded work progress to image hosting
So first I build kernel from source:
1 a) https://ibb.co/PrcMhgN
   b) https://ibb.co/bXdmFvC
   c) https://ibb.co/nssySDS   (Here you can see that the compilation was successful)
2 make
  a) https://ibb.co/NLRQq9X  mrproper
  b) https://ibb.co/r0XT6HZ    defconfig nethunter_defconfig and modules_prepare
  c) https://ibb.co/HLyNZ8b   modules_install and git clone rtl8188eus
3 Next, I configured Makefile
  https://ibb.co/HqxQC77      this
4 And finally "make"
   https://ibb.co/r6t01Pk  this
Hooray, compilation started, but breaks off on 'rtw_mlme_ext.c'
I can’t understand what the problem is, please tell me what I'm doing wrong
My defconfig file if that
http://s000.tinyupload.com/index.php?file_id=12376423205617620599
Thanks

BeingDatta, 2020-05-03:
try with this toolchain

> https://releases.linaro.org/components/toolchain/binaries/latest-7/arm-linux-gnueabihf/gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf.tar.xz

dufduf5, 2020-05-03:
> try with this toolchain
> 
> > https://releases.linaro.org/components/toolchain/binaries/latest-7/arm-linux-gnueabihf/gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf.tar.xz
> 
> Also see this
> 
> > [meekrosoft/fff#32](https://github.com/meekrosoft/fff/issues/32)
> 
> there is something in rtw_mlme_ext.c at line number 1835

Ok, I tried with this toolchain
https://releases.linaro.org/components/toolchain/binaries/latest-7/arm-linux-gnueabihf/gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf.tar.xz
And get it again
https://ibb.co/QnQFx2f
Here is the part of the code rtw_mlme_ext.c on which compilation is interrupted
__#else

	if (GetRetry(pframe)) {
		/* return; */
	}
#endif

#ifdef CONFIG_AP_MODE
	switch (get_frame_sub_type(pframe)) {
	case WIFI_AUTH:
		if (MLME_IS_AP(padapter) || MLME_IS_MESH(padapter))
			ptable->func = &OnAuth;
		else
			ptable->func = &OnAuthClient;
#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 4, 0)
	__attribute__ ((fallthrough));                                               it's line 1835
#else
			__attribute__ ((__fallthrough__));
#endif
	case WIFI_ASSOCREQ:
	case WIFI_REASSOCREQ:
		_mgt_dispatcher(padapter, ptable, precv_frame);
		#ifdef CONFIG_HOSTAPD_MLME
		if (MLME_IS_AP(padapter))
			rtw_hostapd_mlme_rx(padapter, precv_frame);
		#endif
		break;
	case WIFI_PROBEREQ:
		_mgt_dispatcher(padapter, ptable, precv_frame);
		#ifdef CONFIG_HOSTAPD_MLME
		if (MLME_IS_AP(padapter))
			rtw_hostapd_mlme_rx(padapter, precv_frame);
		#endif
as far as I understand, semicolons are necessary here to comply with the syntax
and this is this file 
https://github.com/aircrack-ng/rtl8188eus/blob/v5.3.9/core/rtw_mlme_ext.c

BeingDatta, 2020-05-03:
what version of gcc you are using look at this 

> https://stackoverflow.com/questions/45349079/how-to-use-attribute-fallthrough-correctly-in-gcc

change line 1835 __attribute__ ((fallthrough)); to /* __attribute__ ((fallthrough));  */ that is comment these lines to disable but still I think your gcc version is less than 7.x because  __attribute__ ((fallthrough)); was introduced at gcc 7.x or greater. try and leave comment

dufduf5, 2020-05-03:
Thanks, I'll try. Can I first assemble the kernel with one compiler and then build the module with another? Because the kernel source is not going with the new versions of the compiler and giving a huge amount of errors

BeingDatta, 2020-05-03:
> Thanks, I'll try. Can I first assemble the kernel with one compiler and then build the module with another? Because the kernel source is not going with the new versions of the compiler and giving a huge amount of errors

yes definitely you can do that but remember my notes not to change any name or anything

dufduf5, 2020-05-03:
> > Спасибо попробую. Могу ли я сначала собрать ядро ​​с одним компилятором, а затем собрать модуль с другим? Потому что исходный код ядра не идет с новыми версиями компилятора и выдает огромное количество ошибок
> 
> да, безусловно, вы можете сделать это, но помните мои заметки, чтобы не менять имя или что-либо

I just tried to build the official driver rtl8188eus
https://www.tp-link.com/en/support/download/tl-wn722n/v3/#Driver
And yes, he gathered without any problems...
(By the way, it seems to have monitor mode support)
Hah, but 8188eu.ko does not load into the kernel, I also noticed that in the assembly Kali Nethunter 
there is no such way as /system/lib/modules and correspondingly /lib/modules/$(uname -a)/build too
I thought that the best option would be to build the kernel myself, to collect the headers for it. Then flash the kernel, add the headers and kernel source to the appropriate paths and only then build the modules. Since the official driver is going, it means the problem is in this driver.
I will report on the work done here later

BeingDatta, 2020-05-04:
> I just tried to build the official driver rtl8188eus
> https://www.tp-link.com/en/support/download/tl-wn722n/v3/#Driver
> And yes, he gathered without any problems...
> (By the way, it seems to have monitor mode support)
> Hah, but 8188eu.ko does not load into the kernel, I also noticed that in the assembly Kali Nethunter
> there is no such way as /system/lib/modules and correspondingly /lib/modules/$(uname -a)/build too
> I think that the best option would be to build the kernel myself, to collect the headers for it. Then flash the kernel, add the headers and kernel source to the appropriate paths and only then build the modules. Since the official driver is going, it means the problem is in this driver.
> I will report on the work done here later

yes definitely you have to build the kernel first from source code then only you can do this

kimocoder, 2020-05-04:
There is a slightly easier method to make it build into kernel. @re4son may you provide patch link?

BeingDatta, 2020-05-05:
yes please share it if you have any

Re4son, 2020-05-06:
I don't have any patches for the 8812eus, sorry. It turned out that the driver we've been using never actually worked on Android and I removed it because it caused conflicts with the rtl8812au, which by itself works perfectly.
I recommend to make the driver work properly first and to rename its functions to not conflict with rtl8812au

dufduf5, 2020-05-06:
> There is a slightly easier method to make it build into kernel. @Re4son may you provide patch link?

> I don't have any patches for the 8812eus, sorry. It turned out that the driver we've been using never actually worked on Android and I removed it because it caused conflicts with the rtl8812au, [...]

Yes, it would be cool to add this driver to the kernel
Nethunter, as while the TP-LINK TL-WN722N V2 and V3 adapter is not supported like and some others. But thanks anyway

BeingDatta, 2020-05-06:
what are you telling guys I am using both 8188eu.ko and 88XXau.ko at the same time not conflicting at all

dufduf5, 2020-05-06:
You are talking about rtl8812eu or rtl8188eus, the rtl8188eus driver cannot conflict with rtl8812au if I am not mistaken. I really don’t understand, these are different chips

BeingDatta, 2020-05-07:
Look I want to be clear on this I successfully built two drivers both from aircrack-ng project name rtl8812au and  rtl8188eus. All the guides I provided were only for these two drivers. I never built 8812eus or anything like that if you are trying my guides for building 8812eu or 8812eus then I m sorry mate cant help you in this regard because I dont have any device to test for them. You also should try 8812eus project page if any not here. This place is only for 8188eus.

Regards.

Re4son, 2020-05-07:
Did you build them as kernel modules?
We try to avoid that on Android as the support costs by far outweigh the benefits. I can't really remember which sources I used but Kimocoder pointed me to them and when trying to compile those into the kernel they conflicted because both drivers had identical function names.
I really appreciate your work and I'd love to include all these drivers into NetHunter but they must allow us to build them into the kernel without conflicts.

BeingDatta, 2020-05-07:
yes I build them as kernel modules

BeingDatta, 2020-05-07:
This link https://github.com/aircrack-ng/rtl8188eus/issues/31

Re4son, 2020-05-07:
Nice.
We now ship rtl8812au as part of standard NetHunter kernels. People can come here if they want 8188eus. Sounds like a good deal.  
What do you think?

BeingDatta, 2020-05-07:
yup absolutely why dont you also ship 8188eus along with 8812au ?

Re4son, 2020-05-07:
Because we don't want modules on Android and 8812eus does not compile alongside 8812au.  
Some comedian must have copied half the code from one driver to the other without knowing that global names have to be unique.

BeingDatta, 2020-05-07:
ok now i get that