aircrack-ng/rtl8188eus #50: Compilation for arm64

Si-i3ima, 2020-05-15:
I tried to compile the driver using this toolchain:
For aarch64 : [ https://github.com/kdrag0n/aarch64-elf-gcc.git](url)
For aarch32 :  [https://github.com/kdrag0n/arm-eabi-gcc.git](url)

Here are some lines (which I edited) of the "Platform related" section from the Makefile :
## Platform Related ##

```
CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64_RPI = y
CONFIG_PLATFORM_ANDROID_X86 = n
```
**And a config for arm64:**

```
ifeq ($(CONFIG_PLATFORM_ARM64_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH ?= arm64
CROSS_COMPILE ?= /home/i3ima/toolchains/aarch64/bin/aarch64-elf-
KVER ?= 4.9.223-Dark-Ages-Décimo
KBUILD_EXTMOD ?= /home/i3ima/Projects/kernel_hid/kernel-headers/lib/modules/4.9.223-Dark-Ages-Décimo/source
KSRC := /home/i3ima/Projects/kernel_hid/kernel-headers/lib/modules/4.9.223-Dark-Ages-Décimo/build
MODDESTDIR := /home/i3ima/Projects/kernel_hid/kernel_dark_ages_vince/drivers/net/wireless
INSTALL_PREFIX :=
endif
```

My steps: 
1.  **I assembled the kernel by commands.** 
   
 ```
make O=out vince_defconfig
    make O=out -j9
```
2.  **Assembly and installation of headers**  _(actions were performed in /out directory)._ 
   
 ```
mkdir ../../kernel-headers
    make O=../../kernel-headers vince_defconfig
    make O=../../kernel-headers modules_prepare
    make O=../../kernel-headers modules INSTALL_MOD_PATH=../../kernel-headers 
    make O=../../kernel-headers modules_install INSTALL_MOD_PATH=../../kernel-headers
    make headers_install INSTALL_HDR_PATH=../../kernel-headers 

```

**Errors :** 
```
   CC [M] /home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_rm.o
  CC [M] /home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_rm_fsm.o
  CC [M] /home/i3ima/Projects/kernel_hid/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M] /home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.o
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c: In function 'sdio_io':
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c:56:11: error: 'struct dvobj_priv' has no member named 'intf_ops'.
   56 | err = d->intf_ops->write(d, addr_drv, buf, len, 0);
      | ^~
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c:58:11: error: 'struct dvobj_priv' has no member named 'intf_ops'.
   58 | err = d->intf_ops->read(d, addr_drv, buf, len, 0);
      | ^~
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c:75:56: error: 'SD_IO_TRY_CNT' undeclared (first use in this function)
   75 | if ((err == -1) || (stop_retry == _TRUE) || (retry > SD_IO_TRY_CNT)) {
      | ^~~~~~~~~~~~~
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c:75:56: note: each undeclared identifier is reported only once for each function it appears in
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c: In function 'rtw_sdio_f0_read':
/home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.c:125:9: error: 'struct dvobj_priv' has no member named 'intf_ops'.
  125 | err = d->intf_ops->read(d, addr, buf, len, 0);
      | ^~
make[4]: *** [.../scripts/Makefile.build:340: /home/i3ima/Projects/kernel_hid/rtl8188eus/core/rtw_sdio.o] Error 1
make[3]: *** [/home/i3ima/Projects/kernel_hid/kernel_dark_ages_vince/Makefile:1625: _module_/home/i3ima/Projects/kernel_hid/rtl8188eus]. Error 2
make[2]: *** [Makefile:152: sub-make] Error 2
make[1]: *** [Makefile:24: __sub-make] Error 2
make[1]: Leaving directory '/home/i3ima/Projects/kernel_hid/kernel_dark_ages_vince/out'.
make: *** [Makefile:2048: modules] Error 2 

```

**I hope someone can help me**

drygdryg, 2020-05-28:
Check out my fixes: https://github.com/drygdryg/rtl8188eus
Maybe it will help you.

kimocoder, 2020-05-29:
Check out my patch for kernel 4.4, 4.9 and 4.14
In our build system...

https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-kernel/-/raw/master/patches/4.14/add-rtl88xxau-5.6.4.2-drivers.patch

kimocoder, 2020-05-29:
I have no idea why you guys take the hard way.. it's plain simple, add a pointer from staging Kconfig to driver Kconfig and from staging Makefile to driver Makefile :+1:

Si-i3ima, 2020-05-29:
Thank you for your answer and help.

closed 2020-05-29 by Si-i3ima

kimocoder, 2020-05-29:
I made newer 8188 driver also, in mac80211 subsystem instead on this omd cfg80211/nl80211 mix.

https://github.com/kimocoder/realtek_rtwifi

And our build tool for NetHunter is located below
https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-kernel

Wi-Fi patches, injection.. everything you need :100:

Si-i3ima, 2020-05-29:
> https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-kernel

Is there a patch for kernel 4.9? For HID support?

kimocoder, 2020-05-29:
HID is simple for newer kernels..

Go in the link below and grap the "ramdisk" folder (for anykernel2/3)

Then in NetHunter, setup a custom command, like this:
[image]

Si-i3ima, 2020-05-29:
Thank you, again, I'll try your recommendations.

Si-i3ima, 2020-05-29:
UPD: 
I built the kernel using generic_kernel_patch.4.9.patch to support HID, then copied zImage to the Anykernel folder, and copied the files you specified to the ramdisk folder, then I flashed the resulting archive. I want to ask you a few questions.
First of all, when I run the command "su -c setprop sys.usb.config "win, mtp, hid, adb" two things happen : 
1./dev/hidgX are not generated.
2.in dmesg I see the error "init: Command 'write /config/usb_gadget/g1/UDC none' action=sys.usb.config=none && sys.usb.configfs=1 (/init.usb.configfs.rc:2) took 0ms and failed: cUnable to write to file '/config/usb_gadget/g1/UDC': Unable to write file contents: No such device".
My device has the code name vince, can I use descriptors and initialization scripts from other devices?
Are there instructions on what options menuconfig should include to support HID?

Translated with www.DeepL.com/Translator (free version)

Si-i3ima, 2020-05-30:
Also, i can't set such function
[image]

kimocoder, 2020-05-30:
Did I say you shpuød use a kernel patch? Nope

kimocoder, 2020-05-30:
And init and everything is in the ramdisk folder.

Si-i3ima, 2020-05-30:
I'm sorry for my inattention.I downloaded the kernel sources again, made defconfig for my device and used menuconfig.sh and build.sh to build the kernel and modules, made a zip-archive again, pre-copied the contents of the ramdisk directory into AnyKernel3/ramdisk.

The `svc command usb setFunctions hid` still shows the error "Killed". 
 `setprop usb.config.sys "win, hid, mtp, adb"` also does not give the desired result, mode change simply does not occur

The Arsenal USB section of the NetHunter application shows that I can set the USB function, but nothing happens when you press a button.
[image]

I'll attach my defconfig.
[vince_nh_defconfig.txt](https://github.com/aircrack-ng/rtl8188eus/files/4706784/vince_nh_defconfig.txt)

Si-i3ima, 2020-05-31:
If I doing anything wrong, please tell me, provide guidelines or anything else that can help to solve problem

Si-i3ima, 2020-06-01:
Note :  last time HID was running on 3.18