aircrack-ng/rtl8188eus #51: Evil Twin Attack Not Supported...

websexploiter, 2020-05-26:
**Evil Twin Attack Not Supported**
I just started Evil Twin attack using TP Link TL-WN722N V3 but after so much of research I end up getting driver issues. Please refer to the below error I am getting with Images and codes
`PHY     Interface           Driver          Chipset

null        802.11             ??????          non-mac80211 device? (report this!)
null        dBm                 ??????          non-mac80211 device? (report this!)
null        IEEE                 ??????          non-mac80211 device? (report this!)
null        Mode:Monitor   ??????          non-mac80211 device? (report this!)
null       Tx-Power=13     ??????          non-mac80211 device? (report this!)
            wfphshr-wlan0                          ethtool failed... driver is broken
phy6    wlan0                 8188eu          TP-Link TL-WN722N v2
`

SmillerMP, 2020-05-26:
i have the same problem, with the same adapter. Anyoone have any solution?

adhamkhaled-dev, 2020-06-01:
I had the same problem using the same adapter.
I recommend using Kali in dual boot. You can make the adapter make the deauth attacks while the built in wifi adapter make the fake AP. You can also get another driver that supports that.

SmillerMP, 2020-06-01:
Okay @adhamkhaled-dev  thanks dude

BeingDatta, 2020-06-01:
how u entered mon mode ? is it airmon-ng wlan0 then it wont work u must use ip link and iw dev , i guess then there will be no issue

adhamkhaled-dev, 2020-06-02:
For me, it doesnt work on either way. Are you sure about that?

websexploiter, 2020-06-02:
> I had the same problem using the same adapter.
> I recommend using Kali in dual boot. You can make the adapter make the deauth attacks while the built in wifi adapter make the fake AP. You can also get another driver that supports that.

Bro that's not the concern... I have used it on dual boot only and moreover the main concern is fixing the driver so that we can use that for evil twin attacks too

websexploiter, 2020-06-02:
> how u entered mon mode ? is it airmon-ng wlan0 then it wont work u must use ip link and iw dev , i guess then there will be no issue

I have used that only not the airmon

BeingDatta, 2020-06-02:
I can see yours is v3 mine is v2 working ok donno why yours not doing it right also is your driver version v 5.3.9 becoz its the only one that works ?

websexploiter, 2020-06-02:
Yeah I have used the latest version and my adapter version is 3

blackhatleet96, 2020-07-01:
Mine adaptor v2 but not working HT-WPS breaker or wifite2 non of method work please help me any one have solutions please mail me blackhatleet96@gmail.com