aircrack-ng/rtl8188eus #52: Iwconfig wlan0 mode monitor issue TL-WN722N

vytatuas, 2020-05-27:
[image]

I try to do something about 5-6 hours but I cannot pass to monitor mode. Please help me.

SmillerMP, 2020-05-27:
I think is because you are not super user 

Test with the command 
```
sudo iw dev wlan0 set type monitor
```

Or

Become on super user with the command
```
sudo su 
```

And the correct form tu put your adapter in monitor mode is with the command **iw dev <adaptername> set type monitor**

Read the documentation for more informacion

BeingDatta, 2020-06-19:
use ip link and iw dev instead of ifconfig and iwconfig see the instructions.

Regards

kimocoder, 2020-06-19:
iwconfig is deprecated. Read the docs before opening issue report.

closed 2020-06-19 by kimocoder