aircrack-ng/rtl8188eus #53: problem with airbase-ng and hostapd

SmillerMP, 2020-06-01:
Hello friends, I have been tested using airbase-ng with my TP-WN722N v3 but when I create the AP it won't let me connect to it.

on the other hand I have tried to create the AP with Hostapd but when I start it tells me error in driver 8188eu and it doesn’t work

I do not know if I am doing something wrong, and it is not a problem with the installation of drivers because the injection and the monitor mode work wonderfully.

can anybody help me?  or do you have any solution?  or do you know something?

Thanks for everything

BeingDatta, 2020-06-02:
i guess the driver dosent support ap mode

SmillerMP, 2020-06-02:
☹️ understand, thanks @BeingDatta

SmillerMP, 2020-06-06:
Hi dudes, I wanted to tell you to create a fake AP with an internet connection.  So if you can create AP with TL-WN722N v3 

Good luck

closed 2020-06-06 by SmillerMP