aircrack-ng/rtl8188eus #55: Makefile: Error 2

Luffy-2, 2020-06-05:
I am getting the following error while attempting to run 'sudo make':

```
sudo make
[sudo] password for kali: 
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.5.0-kali2-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: *** /lib/modules/5.5.0-kali2-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2047: modules] Error 2
```

SmillerMP, 2020-06-05:
Hi! You have to install the linux headers 

```
sudo apt-get install linux-headers-$(uname -r)
```

With 👆🏾 comand

Luffy-2, 2020-06-05:
> 
> 
> Hi! You have to install the linux headers
> 
> ```
> sudo apt-get install linux-headers-$(uname -r)
> ```
> 
> With 👆🏾 comand

That command now gives me the following error:

```
sudo apt-get install linux-headers-$(uname -r)
[sudo] password for kali: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-5.5.0-kali2-amd64
E: Couldn't find any package by glob 'linux-headers-5.5.0-kali2-amd64'
E: Couldn't find any package by regex 'linux-headers-5.5.0-kali2-amd64'
```

Luffy-2, 2020-06-05:
Nevermind, got it working by following this guide: https://stackoverflow.com/questions/50759974/unable-to-locate-package-linux-headers-4-15-0-kali2-amd64

closed 2020-06-05 by Luffy-2