aircrack-ng/rtl8188eus #56: At first time installation the monitor mode work after i reboot the monitor mode doesnt work

NewtonOngko, 2020-06-05:
and its get the output like this 
`kali@kali:~/rtl8188eus$ sudo ifconfig wlan0 down
kali@kali:~/rtl8188eus$ sudo iw dev wlan0 set type monitor
command failed: No such device (-19)
`
and then i try to find my driver 
`kali@kali:~/rtl8188eus$ sudo iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
`

I hope some one can help me thanks

Espinosa123321, 2020-06-06:
@NewtonOngko how did you install the driver?

BeingDatta, 2020-06-19:
> and its get the output like this
> `kali@kali:~/rtl8188eus$ sudo ifconfig wlan0 down`
> `kali@kali:~/rtl8188eus$ sudo iw dev wlan0 set type monitor`
> `command failed: No such device (-19)`
>
> and then i [...]

dont use ifconfig and iwconfig use ip link and iw dev instead see instruction.

Regards.

AssafCode, 2021-01-06:
Same problem to me.

botulinski, 2023-06-02:
Heya guys, I have the same problem anyone fixed?

gglluukk, 2023-06-02:
to enable monitor mode run:
```
modprobe 8188eu
sleep 1
ifconfig wlan up
iw dev wlan set type monitor
```

also make sure that default driver for wlan isn't load by:
`ethtool -i wlan`
and if there is remove it:
`rmmod r8188eu`
or
`rmmod rtl8xxxu`