aircrack-ng/rtl8188eus #57: rtw_get_systime_us

Espinosa123321, 2020-06-06:
PLEASE HELP
when I'm going to compile it I always get this error

`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.0-kali2-amd64/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-kali2-amd64'
  CC [M]  /root/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/rtl8188eus/core/rtw_security.o
  CC [M]  /root/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/rtl8188eus/core/rtw_io.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8188eus/core/rtw_ap.o
  CC [M]  /root/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /root/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/rtl8188eus/core/rtw_rson.o
  CC [M]  /root/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /root/rtl8188eus/core/rtw_iol.o
  CC [M]  /root/rtl8188eus/core/rtw_sreset.o
  CC [M]  /root/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /root/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /root/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/rtl8188eus/core/rtw_rm.o
  CC [M]  /root/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /root/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /root/rtl8188eus/os_dep/linux/os_intfs.o
  CC [M]  /root/rtl8188eus/os_dep/linux/usb_intf.o
  CC [M]  /root/rtl8188eus/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/rtl8188eus/os_dep/linux/ioctl_linux.o
  CC [M]  /root/rtl8188eus/os_dep/linux/xmit_linux.o
  CC [M]  /root/rtl8188eus/os_dep/linux/mlme_linux.o
  CC [M]  /root/rtl8188eus/os_dep/linux/recv_linux.o
  CC [M]  /root/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_systime_us’:
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:706:1: warning: no return statement in function returning non-void [-Wreturn-type]
  706 | }
         | ^
  CC [M]  /root/rtl8188eus/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /root/rtl8188eus/os_dep/linux/wifi_regd.o
  CC [M]  /root/rtl8188eus/os_dep/linux/rtw_android.o
  CC [M]  /root/rtl8188eus/os_dep/linux/rtw_proc.o
  CC [M]  /root/rtl8188eus/os_dep/linux/rtw_rhashtable.o
  CC [M]  /root/rtl8188eus/os_dep/linux/ioctl_mp.o
  CC [M]  /root/rtl8188eus/hal/hal_intf.o
  CC [M]  /root/rtl8188eus/hal/hal_com.o
  CC [M]  /root/rtl8188eus/hal/hal_com_phycfg.o
  CC [M]  /root/rtl8188eus/hal/hal_phy.o
  CC [M]  /root/rtl8188eus/hal/hal_dm.o
  CC [M]  /root/rtl8188eus/hal/hal_dm_acs.o
  CC [M]  /root/rtl8188eus/hal/hal_btcoex_wifionly.o
  CC [M]  /root/rtl8188eus/hal/hal_btcoex.o
  CC [M]  /root/rtl8188eus/hal/hal_mp.o
  CC [M]  /root/rtl8188eus/hal/hal_mcc.o
  CC [M]  /root/rtl8188eus/hal/hal_hci/hal_usb.o
  CC [M]  /root/rtl8188eus/hal/led/hal_led.o
[... 402 lines cut ...]
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9427:23: error: initializer element is not constant
 9427 |  .remain_on_channel = cfg80211_rtw_remain_on_channel,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9427:23: note: (near initialization for ‘rtw_cfg80211_ops.remain_on_channel’)
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9428:30: error: initializer element is not constant
 9428 |  .cancel_remain_on_channel = cfg80211_rtw_cancel_remain_on_channel,
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9428:30: note: (near initialization for ‘rtw_cfg80211_ops.cancel_remain_on_channel’)
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9440:13: error: initializer element is not constant
 9440 |  .mgmt_tx = cfg80211_rtw_mgmt_tx,
      |             ^~~~~~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9440:13: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx’)
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9441:25: error: initializer element is not constant
 9441 |  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9441:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_frame_register’)
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9774:1: error: expected declaration or statement at end of input
 9774 | }
      | ^
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9774:1: warning: no return statement in function returning non-void [-Wreturn-type]
make[3]: *** [/usr/src/linux-headers-5.6.0-kali2-common/scripts/Makefile.build:273: /root/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.6.0-kali2-common/Makefile:1704: /root/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.6.0-kali2-common/Makefile:180: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-kali2-amd64'
make: *** [Makefile:2047: modules] Error 2`

azamet90, 2020-06-08:
same problem i do have 
[image]

Espinosa123321, 2020-06-08:
can you make it

Espinosa123321, 2020-06-10:
i already install it, you can take a look inn here https://github.com/kelebek333/rtl8188fu

Espinosa123321, 2020-06-10:
but it doesnt support monitor mode

closed 2020-06-19 by Espinosa123321