aircrack-ng/rtl8188eus #58: RLT8188eus WiFi Usb donggle doesn't want to work...

Scubad1v3r, 2020-06-14:
Hello all,
I am currently at an impasse. I need to configure a second WiFi interface (RLT8188eus USB key) in order to be able to connect an external antenna.
I'm working on :
Linux KaliDiver 5.6.0-kali2-amd64 #1 SMP Debian 5.6.14-1kali1 (2020-05-25) x86_64 GNU/Linux

`lsusb
Bus 003 Device 004: ID 04f2:b160 Chicony Electronics Co., Ltd CNF9113
Bus 003 Device 010: ID 0bda:f179 Realtek Semiconductor Corp. 802.11n
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 25a7:fa61 Compx 2.4G Receiver
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub`

My main WiFi card working perfectly in wlan0
`iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:"Dephdiver_Timeport"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: 10:9A:DD:8A:59:91   
          Bit Rate=65 Mb/s   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=70/70  Signal level=-32 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:2   Missed beacon:0

ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:1f:16:a4:8c:6f brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.17/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 31477203sec preferred_lft 31477203sec
    inet6 fe80::21f:16ff:fea4:8c6f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:22:fb:6f:79:64 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.18/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 31477236sec preferred_lft 31477236sec
    inet6 fe80::944d:9e40:f8ef:10ec/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
 `

The donggle not being recognized, I installed the latest drivers after backlist of the old drivers:
`echo "blacklist r8188eu" > /etc/modprobe.d/realtek.conf
apt-get update
apt-get install realtek-rtl8188eus-dkms`

the driver seems to be installed but no access to the donggle ...
`apt-cache search realtek
flashrom - Identify, read, write, erase, and verify BIOS/ROM/flash chips
librtlsdr-dev - Software defined radio receiver for Realtek RTL2832U (development)
librtlsdr0 - Software defined radio receiver for Realtek RTL2832U (library)
nictools-pci - Diagnostic tools for many PCI Ethernet cards
python3-rtlsdr - Python wrapper for librtlsdr (Python3 package)
rtl-sdr - Software defined radio receiver for Realtek RTL2832U (tools)
soapysdr-module-rtlsdr - RTL-SDR device support for SoapySDR (default version)
soapysdr0.7-module-rtlsdr - RTL-SDR device support for SoapySDR
firmware-realtek - Binary firmware for Realtek wired/wifi/BT adapters
r8168-dkms - dkms source for the r8168 network driver
realtek-rtl8188eus-dkms - Realtek RTL8188EUS driver in DKMS format
realtek-rtl88xxau-dkms - Realtek RTL88xxAU driver in DKMS format`

Firmware seem not to be load
`dmesg | grep firmware
[   22.283437] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[   22.562577] iwlwifi 0000:02:00.0: firmware: direct-loading firmware iwlwifi-5000-5.ucode
[   22.562733] iwlwifi 0000:02:00.0: loaded firmware version 8.83.5.1 build 33692 5000-5.ucode op_mode iwldvm
[   22.705904] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s`

neither the native r8188eu module nor the compiled 8188eu module support donggle
`modinfo -F alias {r,}8188eu | grep 0BDA
usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*
usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*`

I surely forgot to do something, do you have an idea?
Thanks for your help

blackhatleet96, 2020-07-01:
It's your kernel problem change kernel from grub loader andavced its work on 5.3 also many people face this problem coder with update support

Scubad1v3r, 2020-07-01:
> It's your kernel problem change kernel from grub loader andavced its work on 5.3 also many people face this problem coder with update support

Thanks

How to easy downgrade to 5.3?