aircrack-ng/rtl8188eus #59: "make " error2

tanupreet121, 2020-06-15:
Everything was working fine but then i ran apr update && upgrade, after installation make command is showing error 2. What is wrong? Please help. I have TL-WN722N.
root@kali:~# cd rtl8188eus/
root@kali:~/rtl8188eus# sudo -i
root@kali:~# echo "blacklist r8188eu.ko" > "/etc/modprobe.d/realtek.conf"
root@kali:~# exit
logout
root@kali:~/rtl8188eus# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.0-kali4-amd64/build M=/root/rtl8188eus  modules
make[1]: *** /lib/modules/4.19.0-kali4-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2047: modules] Error 2

BeingDatta, 2020-06-19:
install kernel headers google it

kimocoder, 2020-06-19:
What he said. Closing.

closed 2020-06-19 by kimocoder