aircrack-ng/rtl8188eus #60: TL-WN722N adapter not detecting in Kali kernel 5.6.0 (latest kernel)

Mukundhan98, 2020-06-20:
I'm having a trouble connecting my adapter in Kali VM kernel 5.6.0. I recently upgraded my Kali to latest version, which changes my kali kernel from 5.3.0 to 5.6.0. My adapter in not being detected in kernel 5.6.0.

I tried changing the kernel from 5.6.0 to 5.3.0 using grub loader advanced and my adapter is working fine in kernel 5.3.0.

So I have to change to kernel 5.3.0 every time. This is annoying. Does anyone have a solution?

blackhatleet96, 2020-07-01:
Ya but already kimocoder added kernel support 5.6 nd this adaptor is working fine in kali did you goted 2-4wifi connection ???

Mukundhan98, 2020-07-02:
> Ya but already kimocoder added kernel support 5.6 nd this adaptor is working fine in kali did you goted 2-4wifi connection ???

I am using this adapter for Network testing. Is there any ways or steps to install latest driver which support kernel 5.6?