aircrack-ng/rtl8188eus #63: Cannot change MAC Address on Kali

notsars, 2020-06-29:
Working with a Wireless adapter (TL-WN725N), which I managed to configure on my VM Kali. However, I cannot change the MAC Address.

I have tried everything:
ifconfig wlan0 down
ifconfig wlan0 hw ether 00:11:22:22:33:44
Ifconfig wlan0 up

ifconfig wlan0 down
macaddress -m 00:00:00:00:00:00
ifconfigu wlan0 up

Any suggestions?

SmillerMP, 2020-06-30:
Hi @notsars  Can you post any screenshot about the result of that comand?

notsars, 2020-06-30:
root@kali:~# macchanger -r wlan0
Current MAC:   XX:XX:XX:XX:XX:XX (unknown)
Permanent MAC: XX:XX:XX:XX:XX:XX (unknown)
New MAC:       XX:XX:XX:XX:XX:XX (unknown)
Network driver didn't actually change to the new MAC!!

This is what keeps on appearing. Otherwise, when I use the above commands, it just continues the same as the current MAC.

SmillerMP, 2020-07-01:
Can you post the result about the ```
iwconfig 
```comand?

notsars, 2020-07-01:
eth0      no wireless extensions.

lo        no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

SmillerMP, 2020-07-01:
why better reinstall the drivers from scratch and be sure to follow the steps for proper installation.

BeingDatta, 2020-07-14:
> root@kali:~# macchanger -r wlan0
> Current MAC:   XX:XX:XX:XX:XX:XX (unknown)
> Permanent MAC: XX:XX:XX:XX:XX:XX (unknown)
> New MAC:       XX:XX:XX:XX:XX:XX (unknown)
> Network driver didn't actually [...]

probably you missed the very first instruction to blacklist the built in driver first or do one thing after blacklisting do a dkms install.

Regards