aircrack-ng/rtl8188eus #64: Make error

KedarJoshi11, 2020-07-02:
Hi

After cloning, when I run the make && make install command, it throws the following error. 

/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-kali3-amd64/build M=/root/rtl8188eus  modules
make[1]: *** /lib/modules/5.4.0-kali3-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2047: modules] Error 2

Please help. This Wireless LAN isn't even showing up when I do ifconfig. My other adapter of Netgear is working fine. 

[image]

KTM888, 2020-07-02:
same here: 
[image]

SmillerMP, 2020-07-03:
You have to install the linux headers

```
sudo apt-get install linux-headers-$(uname -r)
````

KedarJoshi11, 2020-07-03:
> You have to install the linux headers
> 
> ```
> sudo apt-get install linux-headers-$(uname -r)
> ```

[image]

See this, can you help me find headers for this version?

KedarJoshi11, 2020-07-03:
> same here:
> [image]

Can you tell me which wireless adapter you are using?

SmillerMP, 2020-07-03:
@KedarJoshi11  Try with this comand

```
sudo apt-get install build-essential linux-headers-$(uname -r)

```

KedarJoshi11, 2020-07-03:
> @KedarJoshi11 Try with this comand
> 
> ```
> sudo apt-get install build-essential linux-headers-$(uname -r)
> ```
> 
> I use the tl-wn722n v3, but I think it has nothing to do with the adapter

@SmillerMP 

I used the command you suggested. Still the same result, how do you think this can be resolved?

KedarJoshi11, 2020-07-03:
> @KedarJoshi11 Try with this comand
> 
> ```
> sudo apt-get install build-essential linux-headers-$(uname -r)
> ```
> 
> I use the tl-wn722n v3, but I think it has nothing to do with the adapter

[image]

see this.

KedarJoshi11, 2020-07-03:
5.4.0-kali3-amd64

I need to find a header file for this version.

SmillerMP, 2020-07-03:
can you install programs?

Try to install screenfetch

```
sudo apt-get install screenfetch
``` 

And post the result

KTM888, 2020-07-03:
> > same here:
> > [image]
> 
> Can you tell me which wireless adapter you are using?

I am using the TL-WN722N v2

KTM888, 2020-07-03:
> > > same here:
> > > [image]
> > 
> > 
> > Can you tell me which wireless adapter you are using?
> 
> I am using the TL-WN722N v2

[image]

KTM888, 2020-07-03:
> > > same here:
> > > [image]
> > 
> > 
> > Can you tell me which wireless adapter you are using?
> 
> I am using the TL-WN722N v2

For Arch Linux users use ```code sudo pacman -S linux56-headers``` for the latest 5.6 headers

SmillerMP, 2020-07-03:
@KTM888  

sorry I don't know that much about Arch Linux

SmillerMP, 2020-07-03:
@KedarJoshi11  @KTM888  You have to search the linux headers for your system

KTM888, 2020-07-03:
the headers work fine but, when i try to set it to monitor mode i got an error 
[image]

SmillerMP, 2020-07-03:
You must to put your adapter on monitor mode with the comands:

```
airmon-ng check kill
ip link set (adapter name) down
iw dev (adapter name) set type monitor

```

KedarJoshi11, 2020-07-03:
> @KedarJoshi11 @KTM888 You have to search the linux headers for your system

Yes, I know. But I can't find them at the moment the version I am using apparently there are none. I can't find them on the internet at the moment. 

uname -r gives this result. 

5.4.0-kali3-amd64

Can't find headers for this even on that long list of page where headers for different versions are present.

murprakoso, 2020-07-09:
> > > > same here:
> > > > [image]
> > > 
> > > 
> > > Can you tell me which wireless adapter you are using?
> > 
> > 
> > I am using the TL-WN722N v2
> 
> For Arch Linux users use ` sudo pacman -S linux56-headers` for the latest 5.6 headers

not work for me :( , 5.4.50-1-lts kernel

KedarJoshi11, 2020-07-10:
I couldn't find the header files for the my kernel version either. Let me
know if you find something that resolves this please.

SmillerMP, 2020-07-13:
Hi guys, the problem was fixed, thanks @kimocoder

KedarJoshi11, 2020-07-13:
How, can you tell me as well? My wireless adapter is still not working?

SmillerMP, 2020-07-13:
Clone and make again!

KedarJoshi11, 2020-07-13:
> Clone and make again!

Did this on a completely new installation of Kali 2020.2a. But can't install Linux headers for this one either. Meanwhile, my other Netgear Wireless adapter works perfectly fine on the same system. 

[image]

kimocoder, 2020-07-13:
You do not have "bc": apt-get install bc
U use " infrodnt of echo? why?

KedarJoshi11, 2020-07-13:
> You do not have "bc": apt-get install bc
> U use " infrodnt of echo? why? --> 

Hi Christian, 

I accidently copied it from the instructions with the quotation mark, then I realized that I already did that before. Anyway, I did apt-get install bc, but I am still getting this error. 

[image]

kimocoder, 2020-07-13:
Missing the most essential, kernel headers.
apt-cache search linux-headers

KedarJoshi11, 2020-07-14:
> Missing the most essential, kernel headers.
> apt-cache search linux-headers

I used this command and then did make && make install. Still the same error. 

[image]

Tymofei, 2020-07-15:
try sudo apt-get install bc

InstanXXteternal, 2020-07-16:
you can find linux-headers in 
http://http.kali.org/kali/pool/main/l/linux/

barrett092, 2020-07-24:
also receiving this error - brutal

venken, 2020-07-27:
me too has this prolem ,
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.7.0-kali1-amd64/build M=/home/kali/Desktop/rtl8188eus/rtl8188eus  modules
make[1]: Entering directory '/usr/lib/modules/5.7.0-kali1-amd64/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.7.0-kali1-amd64/build'
make: *** [Makefile:2313: modules] Error 2
i installed header

SmillerMP, 2020-07-27:
first run `apt update`

 then can you show me the output of the following commands 

1. `uname -r`
2. `apt-cache search linux-image`

preferably show full command output

mobi35, 2020-08-10:
Hi I'm also getting this error

[image]

I already have bc and installed linux headers

[image]

this is my uname -r

[image]

screenfetch

[image]

what do you think is wrong ? I'm trying to downgrade my kernel but I dont know how. I'm kinda new to linux. Help please.

SmillerMP, 2020-08-11:
@mobi35 I think the kernel 5.7 it isn't compatible
the kernel can´t downgrade 
try with other version of Kali, i use the version 2019.1 and works good

anshulbansal02, 2020-08-14:
Kernel 5.7 is working fine for me on WN722N V2.  I also encounter similar error so I upgraded my kernel using `sudo apt-get update && sudo apt-get upgrade`. After that installed the linux headers and I was able to put my adapter to monitor mode. But there is some abnormal behaviour, like it is not capturing packets as it should be. As it is only capturing a few packets from my own access point even when a lot of devices are connected to it and are not idle.

Mr-Tintin, 2020-08-14:
i have a problem wen i put make in the terminal i give me an erreu n°2

[image]
if someone can help me pls.

anshulbansal02, 2020-08-14:
@Mr-Tintin , you have to install linux-headers package by running command. `sudo apt-get update && sudo apt-get install linux-headers-$(uname -r)` . If you are not able to install it then first you have to upgrade your kernel to the latest kernel available to install that package.

Mr-Tintin, 2020-08-14:
[image]

SmillerMP, 2020-08-14:
@Mr-Tintin  first run `apt update`

 then can you show me the output of the following commands 

1. `uname -r`
2. `apt-cache search linux-image`
3. `apt-cache search linux-headers`

preferably show full command output

Mr-Tintin, 2020-08-14:
[image]

SmillerMP, 2020-08-14:
@Mr-Tintin  Man what is the result of 

```
apt-cache search linux-image
```
```
apt-cache search linux-headers
```

Mr-Tintin, 2020-08-14:
For image IS :
linux-headers-5.7.0-kali1-amd64 - Header files for Linux 5.7.0-kali1-amd64
linux-headers-5.7.0-kali1-cloud-amd64 - Header files for Linux 5.7.0-kali1-cloud-amd64
linux-image-5.7.0-kali1-amd64 - Linux 5.7 for 64-bit PCs
linux-image-5.7.0-kali1-amd64-dbg - Debug symbols for linux-image-5.7.0-kali1-amd64
linux-image-5.7.0-kali1-cloud-amd64 - Linux 5.7 for x86-64 cloud
linux-image-5.7.0-kali1-cloud-amd64-dbg - Debug symbols for linux-image-5.7.0-kali1-cloud-amd64
linux-image-amd64 - Linux for 64-bit PCs (meta-package)
linux-image-amd64-dbg - Debugging symbols for Linux amd64 configuration (meta-package)
linux-image-cloud-amd64 - Linux for x86-64 cloud (meta-package)
linux-image-cloud-amd64-dbg - Debugging symbols for Linux cloud-amd64 configuration (meta-package)
linux-image-5.5.0-kali2-amd64 - Linux 5.5 for 64-bit PCs
linux-image-5.6.0-kali1-amd64 - Linux 5.6 for 64-bit PCs

And
For  headers i have nothing

SmillerMP, 2020-08-14:
@Mr-Tintin Okay, now try to this

`sudo apt install linux-headers-5.7.0-kali1-amd64`

Then reboot and post the results of command `apt-cache search linux-headers`

Mr-Tintin, 2020-08-14:
linux-headers-5.7.0-kali1-amd64 - Header files for Linux 5.7.0-kali1-amd64
linux-headers-5.7.0-kali1-cloud-amd64 - Header files for Linux 5.7.0-kali1-cloud-amd64
linux-image-5.7.0-kali1-amd64 - Linux 5.7 for 64-bit PCs
linux-image-5.7.0-kali1-amd64-dbg - Debug symbols for linux-image-5.7.0-kali1-amd64
linux-image-5.7.0-kali1-cloud-amd64 - Linux 5.7 for x86-64 cloud
linux-image-5.7.0-kali1-cloud-amd64-dbg - Debug symbols for linux-image-5.7.0-kali1-cloud-amd64
linux-image-amd64 - Linux for 64-bit PCs (meta-package)
linux-image-amd64-dbg - Debugging symbols for Linux amd64 configuration (meta-package)
linux-image-cloud-amd64 - Linux for x86-64 cloud (meta-package)
linux-image-cloud-amd64-dbg - Debugging symbols for Linux cloud-amd64 configuration (meta-package)
linux-image-5.5.0-kali2-amd64 - Linux 5.5 for 64-bit PCs
linux-image-5.6.0-kali1-amd64 - Linux 5.6 for 64-bit PCs

And After reboot again nothing

SmillerMP, 2020-08-14:
didi you run `sudo apt install linux-headers-5.7.0-kali1-amd64`  ????...

SmillerMP, 2020-08-14:
@Mr-Tintin 
You can install version 5.7 of the kernel and thus obtain the headers for version 5.7, it is your decision if you want to do it.

if you want to do: run the command `sudo apt install linux-image-5.7.0-kali1-amd64 `
then reboot
and then try to install the headers `sudo apt install linux-headers-5.7.0-kali1-amd64` 
or `sudo apt-get install linux-headers-$(uname -r)`

comment if it worked for you

Mr-Tintin, 2020-08-14:
I want to upgrade kernel but m'y kali dont want

SmillerMP, 2020-08-14:
@Mr-Tintin lol, what happend There are any error?

Mr-Tintin, 2020-08-14:
THIS SHIT 

root@kali:/home/redbear# sudo apt-get install linux-headers-5.7.0-kali1-amd64
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.7.0-kali1-amd64 is already the newest version (5.7.6-1kali2).
The following packages were automatically installed and are no longer required:
  libkf5iconthemes-bin libmpdec2 libpython3.7-minimal libpython3.7-stdlib python3.7 python3.7-minimal qml-module-org-kde-runnermodel vlc-l10n vlc-plugin-notify
  vlc-plugin-samba vlc-plugin-video-splitter vlc-plugin-visualization
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 311 not upgraded.

SmillerMP, 2020-08-14:
@Mr-Tintin hey man you already have the updated version of the headers, what is your version of kernel?

Mr-Tintin, 2020-08-14:
How see this ? Im a beginner

SmillerMP, 2020-08-14:
@Mr-Tintin why better not give me your twitter account? it would be better:))

Mr-Tintin, 2020-08-14:
Why not but i dont realy use him

https://twitter.com/Descieux3?s=09

SmillerMP, 2020-08-14:
@Mr-Tintin or run `uname -r` and post the result

Mr-Tintin, 2020-08-14:
She Say 5.6.0-kali1-amd64

SmillerMP, 2020-08-14:
@Mr-Tintin   try `sudo apt install linux-image-5.7.0-kali1-amd64 `

Mr-Tintin, 2020-08-14:
Do you have discorde ?

Mr-Tintin, 2020-08-14:
And i think it the commande work
[image]

SmillerMP, 2020-08-15:
@Mr-Tintin  no man, i don have, and i dont speak english very well, sorry.

Okay now try to reboot and make de rtl8188eus and say me it this work

Mr-Tintin, 2020-08-15:
Me to im french boyyyyy and thanks you X1000 i make m'y self After kranel script you give me and finaly work FCK WELLL

[110 more replies]