aircrack-ng/rtl8188eus #65: Make Error: No rule to make target "arch//Makefile"

xXAnToS12Xx, 2020-07-08:
Title says everything:
[image]

kiostudio, 2020-07-09:
Facing the same issue here, anyone can help ? 
[image]

murprakoso, 2020-07-10:
[image]

same issue :(

drygdryg, 2020-07-12:
Latest commit https://github.com/aircrack-ng/rtl8188eus/commit/180a406396ab753c1d41e1aa9180eb64cdba9724 breaks build on Arch Linux (x86_64) and Linux Mint (x86):
```
victor:~/git/rtl8188eus $ make -j3
make ARCH= CROSS_COMPILE= -C /lib/modules/5.7.7-arch1-1/build M=/home/victor/git/rtl8188eus  modules
make[1]: Entering directory «/usr/lib/modules/5.7.7-arch1-1/build»
Makefile:669: arch//Makefile: No such file or directory
make[1]: *** No rule to make target «arch//Makefile».  Stop.
make[1]: Leaving directory  «/usr/lib/modules/5.7.7-arch1-1/build»
make: *** [Makefile:2058: modules] Error 2
```

kimocoder, 2020-07-12:
Fixed. git pull and build

closed 2020-07-12 by kimocoder