aircrack-ng/rtl8188eus #67: Kernel 5.7.9: ioctl_cfg80211 Error

AngryPhantom, 2020-07-16:
Hello, guys

Anyone using the latest kernel? Mine got updated from 5.4.51 to 5.7.9 and now the module fails to compile. Here is the [make.log](https://pastebin.com/raw/JLVcM6Ff).

My device is TL-WN722N (V3) and everything was fine on previous kernels.

Any thoughts? Should I try the 5.7.6.1 branch of the driver?
Btw, as I can see (probably a similar issue?) someone has suggested a [patch](https://github.com/aircrack-ng/rtl8188eus/issues/41) here. Should I try this too?

Thanks.

sanyam8929, 2020-07-27:
if you find any solution pls let me know because from last 3 days i am searching and getting same error

AngryPhantom, 2020-07-27:
I used the 5.7.6.1 branch with the patch mentioned above and it works with
no issues for me. But the original code should be fixed and updated.

LevitatingBusinessMan, 2020-08-20:
Are we talking about
```
/home/rein/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9435:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
```
?

AngryPhantom, 2020-08-20:
@LevitatingBusinessMan Nope. There is a link to the [make.log](https://pastebin.com/raw/JLVcM6Ff) in my post.

sndnvaps, 2023-03-25:
@LevitatingBusinessMan i have the same problem ,build with kernel 5.4.188,