aircrack-ng/rtl8188eus #68: getting error in the new kali linux version

sanyam8929, 2020-07-27:
this patch was working fine in the older verion but it is noy working in the new version ..
so pls give new patch for the drivers

amir1376irani, 2020-07-29:
@sanyam8929 It will make processing your issue faster if you say what's your Kali version, and what you mean by older version.
Not only in this repository but also in other Github issues.

sanyam8929, 2020-07-29:
> @sanyam8929 It will make processing your issue faster if you say what's your Kali version, and what you mean by older version.
> Not only in this repository but also in other Github issues.

this is my version
 Linux kali 5.7.0-kali1-amd64 #1 SMP Debian 5.7.6-1kali2 (2020-07-01) x86_64 GNU/Linux