aircrack-ng/rtl8188eus #72: TL-WN722N_V3.0 With RTL8188EUS chipset. It supports Monitor mode and packet injection  or not?!

zankoatzanko, 2020-08-15:
Hello.
I had a question, I want to buy the **TL-WN722N_V3.0** wireless network card, which is from the **RTL8188EUS** chipset. I wanted to know if it really supports Monitor mode and packet injection  or not ??
Thanks a lot.

SmillerMP, 2020-08-15:
Hi. I have the same version of adapter! And it works fantastic! 

It support monitor mode and frame injection very well!

zankoatzanko, 2020-08-15:
> Hi @zankoatzanko I have the same version of adapter! And it works fantastic!
> 
> It support monitor mode and frame injection very well!

Thanks so much for the quick reply.

closed 2020-08-15 by zankoatzanko

resist15, 2020-08-17:
Does It Works With Fluxion Or Hostapd

SmillerMP, 2020-08-17:
@resist15 I think that with fluxion it is not compatible and with hostapd yes but if you configure it correctly

drygdryg, 2020-10-15:
@resist15 it works with fluxion well due to support AP and monitor modes.