aircrack-ng/rtl8188eus #73: Build error: redefinition of ‘struct sha256_state’ (kernel 5.8.2)

LevitatingBusinessMan, 2020-08-19:
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.2-050802-generic/build M=/home/rein/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.2-050802-generic'
  CC [M]  /home/rein/rtl8188eus/core/rtw_cmd.o
In file included from /home/rein/rtl8188eus/include/drv_types.h:77,
                 from /home/rein/rtl8188eus/core/rtw_cmd.c:17:
/home/rein/rtl8188eus/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /home/rein/rtl8188eus/include/osdep_service_linux.h:48,
                 from /home/rein/rtl8188eus/include/osdep_service.h:50,
                 from /home/rein/rtl8188eus/include/drv_types.h:26,
                 from /home/rein/rtl8188eus/core/rtw_cmd.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:281: /home/rein/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1756: /home/rein/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.2-050802-generic'
make: *** [Makefile:2059: modules] Error 2
```

uname -a: `Linux pop-os 5.8.2-050802-generic #202008190732 SMP Wed Aug 19 07:35:42 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

SmartBoy84, 2020-08-24:
I am also having the same problem when building the kernel.
Using the 5.8.1-surface linux kernel
https://github.com/linux-surface/linux-surface

SmartBoy84, 2020-08-29:
I was able to resolve this issue and have sent a pull request but until it is approved you can use my fork:
https://github.com/SmartBoy84/rtl8188eus

kimocoder, 2020-08-29:
Problem should be resolved

closed 2020-08-29 by kimocoder

TsXor, 2022-02-14:
> Problem should be resolved

In which commit did you solve it? How?