aircrack-ng/rtl8188eus #76: sha build error on kernel-5.8... gcc=10.2.0

ohquait, 2020-08-28:
DKMS make.log for rtl8188eus-5.7.6.1-3pclos2020 for kernel 5.8.5-pclos1 (x86_64)
Fri 28 Aug 2020 09:19:28 AM UTC
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.5-pclos1/build M=/var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build  modules
make[1]: Entering directory '/usr/src/kernel-devel-5.8.5-pclos1'
  CC [M]  /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/include/drv_types.h:75,
                 from /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/core/rtw_cmd.c:17:
/var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
  252 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/linux/tcp.h:19,
                 from ./include/linux/ipv6.h:88,
                 from ./include/net/addrconf.h:50,
                 from /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/include/osdep_service_linux.h:48,
                 from /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/include/osdep_service.h:50,
                 from /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/core/rtw_cmd.c:17:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1747: /var/lib/dkms/rtl8188eus/5.7.6.1-3pclos2020/build] Error 2
make[1]: Leaving directory '/usr/src/kernel-devel-5.8.5-pclos1'
make: *** [Makefile:2316: modules] Error 2

drygdryg, 2020-08-28:
I am also seeing issues with the build of version 5.7.6 on new Linux kernels. Use the version 5.3.9

SmartBoy84, 2020-08-29:
> I am also seeing issues with the build of version 5.7.6 on new Linux kernels. Use the version 5.3.9

#73

closed 2020-08-29 by kimocoder