aircrack-ng/rtl8188eus #81: /bin/sh: 1: bc: not found

dickyharper, 2020-09-17:
I need help. I don't know what to do. 

My wireless card: Tp-Link tl-wn727n v5.20 (r8188eu)

Kali version in VirtualBox: Kali_linux_5.5.0_2020_2a_vbox.ova 

'root@kali:/home/kali/Desktop/rtl8188eus-5.3.9# make && make install
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.5.0-kali2-amd64/build M=/home/kali/Desktop/rtl8188eus-5.3.9  modules
make[1]: Entering directory '/usr/src/linux-headers-5.5.0-kali2-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_cmd.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_security.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_debug.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_io.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_ioctl_query.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_ioctl_set.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_ieee80211.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_mlme.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_mlme_ext.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_mi.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_wlan_util.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_vht.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_pwrctrl.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_rf.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_recv.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_sta_mgt.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_ap.o
/home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_ap.c:195:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  195 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/mesh/rtw_mesh.o                                                                                                             
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/mesh/rtw_mesh_pathtbl.o                                                                                                     
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_xmit.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_p2p.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_rson.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_tdls.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_br_ext.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_iol.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_sreset.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_btcoex_wifionly.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_btcoex.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_beamforming.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_odm.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_rm.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/rtw_rm_fsm.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/core/efuse/rtw_efuse.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/osdep_service.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/os_intfs.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/usb_intf.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/ioctl_linux.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/xmit_linux.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/mlme_linux.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/recv_linux.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/wifi_regd.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/rtw_android.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/rtw_proc.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/rtw_rhashtable.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/os_dep/linux/ioctl_mp.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/hal/hal_intf.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/hal/hal_com.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/hal/hal_com_phycfg.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/hal/hal_phy.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/hal/hal_dm.o
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/hal/hal_dm_acs.o
[... 74 lines cut ...]
  CC [M]  /home/kali/Desktop/rtl8188eus-5.3.9/8188eu.mod.o
  LD [M]  /home/kali/Desktop/rtl8188eus-5.3.9/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.5.0-kali2-amd64'
/bin/sh: 1: bc: not found
install -p -m 644 8188eu.ko  /lib/modules/5.5.0-kali2-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.5.0-kali2-amd64
root@kali:/home/kali/Desktop/rtl8188eus-5.3.9# root@kali:/home/kali/Desktop/rtl8188eus-5.3.9# make && make install
bash: root@kali:/home/kali/Desktop/rtl8188eus-5.3.9#: No such file or directory
root@kali:/home/kali/Desktop/rtl8188eus-5.3.9# /bin/sh: 1: bc: not found
bash: /bin/sh:: No such file or directory'

'root@kali:~# apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.5.0-kali2-amd64 is already the newest version (5.5.17-1kali1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.'

'root@kali:~# apt-get install build-essential linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.8).
linux-headers-5.5.0-kali2-amd64 is already the newest version (5.5.17-1kali1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.'

kimocoder, 2020-09-17:
apt-get install bc

closed 2020-09-17 by kimocoder

Bukhosie, 2020-09-17:
Currently facing the same problem ,when I first connect my 722n ,it connects then later on during pen testing process commands ,it just stops working .
Any one with the solution or something ,I've tried on all usb settings but still the same problem .

dickyharper, 2020-09-17:
> apt-get install bc

root@kali:~# apt-get install bc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package bc
root@kali:~#

dickyharper, 2020-09-18:
> apt-get install bc

Yes it worked.  The driver was installed without an error message.  But now Kali doesn't see my wireless card at all.  Is there a solution to this? 

' root@kali:/home/kali# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

root@kali:/home/kali# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.2.15  netmask 255.255.255.0  broadcast 10.0.2.255
        inet6 fe80::a00:27ff:fe23:ff90  prefixlen 64  scopeid 0x20<link>
        ether 08:00:27:23:ff:90  txqueuelen 1000  (Ethernet)
        RX packets 6  bytes 930 (930.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 29  bytes 2443 (2.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 16  bytes 796 (796.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 16  bytes 796 (796.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@kali:/home/kali# airmon-ng

PHY     Interface       Driver          Chipset '