aircrack-ng/rtl8188eus #83: make & make install error kali 2020

SamirMahmudzade, 2020-09-21:
uname -a
Linux kali 5.7.0-kali1-amd64 #1 SMP Debian 5.7.6-1kali2 (2020-07-01) x86_64 GNU/Linux

kali@kali:~/rtl8188eus$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.7.0-kali1-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: *** /lib/modules/5.7.0-kali1-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2059: modules] Error 2

Kali version 2020.3

when i run make command show me error.

Please Help me.

SmillerMP, 2020-09-21:
hi @SamirMahmudzade 
you have to install the linux headers:
`sudo apt-get install linux-headers-$(uname -r)`
then make

SamirMahmudzade, 2020-09-24:
> 
> 
> hi @SamirMahmudzade
> you have to install the linux headers:
> `sudo apt-get install linux-headers-$(uname -r)`
> then make

i try. but same problem.
after upgrade fixed problem

SamirMahmudzade, 2020-09-24:
do you know about wifite ?
because monitor mode work without problem but when i run wifite show this >

# wifite
   .               .    
 .´  ·  .     .  ·  `.  wifite2 2.5.5
 :  :  :  (¯)  :  :  :  a wireless auditor by @derv82
 `.  ·  ` /¯\ ´  ·  .´  maintained by kimocoder
   `     /¯¯¯\     ´    https://github.com/kimocoder/wifite2

 [!] Warning: Recommended app pyrit was not found. install @ https://github.com/JPaulMora/Pyrit/wiki
 [!] Warning: Recommended app hcxdumptool was not found. install @ https://github.com/ZerBea/hcxdumptool
 [!] Warning: Recommended app hcxpcaptool was not found. install @ https://github.com/ZerBea/hcxtools

    Interface   PHY   Driver              Chipset                                                                          
-----------------------------------------------------------------------
 1. wlan0       phy4  8188eu              TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]

 [+] enabling monitor mode on wlan0... failed
failed

 [!] Error: Cannot find any interfaces in monitor mode
                                                                                                                           
 [!] Full stack trace below                                                                                                
                                                                                                                           
 [!]    Traceback (most recent call last):                                                                                 
 [!]    File "/usr/lib/python3/dist-packages/wifite/__main__.py", line 97, in entry_point
 [!]        wifite.start()
 [!]    File "/usr/lib/python3/dist-packages/wifite/__main__.py", line 51, in start                                        
 [!]        Configuration.get_monitor_mode_interface()
 [!]    File "/usr/lib/python3/dist-packages/wifite/config.py", line 164, in get_monitor_mode_interface                    
 [!]        cls.interface = Airmon.ask()
 [!]    File "/usr/lib/python3/dist-packages/wifite/tools/airmon.py", line 330, in ask                                     
 [!]        iface.interface = Airmon.start(iface)
 [!]    File "/usr/lib/python3/dist-packages/wifite/tools/airmon.py", line 204, in start                                   
 [!]        raise Exception('Cannot find any interfaces in monitor mode')
 [!]  Exception: Cannot find any interfaces in monitor mode                                                                
                                                                                                                           
 [!] Exiting

kiryosha, 2020-09-30:
> > hi @SamirMahmudzade
> > you have to install the linux headers:
> > `sudo apt-get install linux-headers-$(uname -r)`
> > then make
> 
> i try. but same problem.
> after upgrade fixed problem

> > hi @SamirMahmudzade
> > you have to install the linux headers:
> > `sudo apt-get install linux-headers-$(uname -r)`
> > then make
> 
> i try. but same problem.
> after upgrade fixed problem

How fix?

N3rfv, 2020-11-04:
i'm sorry for my english the problem is in the version on linux headers, the patch the driver is for headers 5.6 and you have de headers 5.7 so i recommend you downground the headers because i had the same problem, i had the headers 5.8.14 for the kali linux version 2020.3

Miha22, 2021-05-21:
> hi @SamirMahmudzade
> you have to install the linux headers:
> `sudo apt-get install linux-headers-$(uname -r)`
> then make

This works

closed 2022-03-19 by kimocoder