aircrack-ng/rtl8188eus #86: monitor mode not working: Raspberry Pi, Kali and TP-LINK TL-WN722N (EU) Version 3.20

TheBallOfAeolus, 2020-09-30:
Hi everybody
I have a Raspberry Pi 4 Model B Rev 1.2 and a TP-LINK TL-WN722N (EU) Ver.:3.20 (150 Mbps High Gain Wireless USB Adapter - White)

Installed:
OS: Kali Linux
Kernel: aarch64 Linux 4.19.118-Re4son-v8l+
CPU: BCM2835 @ 4x 1.5GHz

USB:
```
:~$ sudo lshw -C network
  *-network:1
       description: Wireless interface
       physical id: 2
       bus info: usb@1:1.4
       logical name: wlan0
       serial: d0:37:45:10:aa:bd
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=r8188eu multicast=yes wireless=unassociated
```

Compiling:
```
:~$ git clone https://github.com/aircrack-ng/rtl8188eus
Cloning into 'rtl8188eus'...
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (57/57), done.
remote: Total 2056 (delta 20), reused 26 (delta 7), pack-reused 1991
Receiving objects: 100% (2056/2056), 5.51 MiB | 9.31 MiB/s, done.
Resolving deltas: 100% (955/955), done.

:~$ cd rtl8188eus/

:~$ make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v8l+/build M=/home/username/dev/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v8l+'
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_security.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_io.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ap.o
/home/username/dev/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/home/username/dev/rtl8188eus/core/rtw_ap.c:195:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  195 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rson.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex.o
[... 162 lines cut ...]
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
drm                   512000  0
drm_panel_orientation_quirks    20480  1 drm
ip_tables              28672  0
x_tables               40960  1 ip_tables
ipv6                  507904  50
```

...but monitor mode is not working! :(
```
:~$ sudo ifconfig wlan0 down
:~$ sudo airmon-ng check kill

Killing these processes:

    PID Name
    384 dhclient
    455 wpa_supplicant

:~$ sudo iw dev wlan0 set type managed
command failed: No such device (-19)
```

Any suggestion?

VINAYNIC, 2020-10-10:
Run airmon-ng before ifconfig wlan0 down and then try

kimocoder, 2020-10-10:
U know your cloning the 'master' branch and not the v5.3.9, right?

TheBallOfAeolus, 2020-10-11:
> U know your cloning the 'master' branch and not the v5.3.9, right?

I wasn't aware I had to use v5.3.9.
I started from scratch using the branch v5.3.9 and now I am getting an error during `make` .
I have created a different issue for it: [Error 91](https://github.com/aircrack-ng/rtl8188eus/issues/91)

TheBallOfAeolus, 2020-10-11:
> Run airmon-ng before ifconfig wlan0 down and then try

Hi @VINAYNIC 
I tried your suggestion, but I am getting the same result
```
username@kali:~$ sudo airmon-ng check kill
Killing these processes:

    PID Name
    387 dhclient
    455 wpa_supplicant

username@kali:~$ sudo ifconfig wlan0 down

username@kali:~$ sudo iw dev wlan0 set type managed
command failed: No such device (-19)
```

TheBallOfAeolus, 2020-10-13:
> > U know your cloning the 'master' branch and not the v5.3.9, right?
> 
> Hi @kimocoder , I wasn't aware I had to use v5.3.9.
> I started from scratch using the branch v5.3.9 and now I am getting an error during `make` .
> I have created a different issue for it: [Error 91](https://github.com/aircrack-ng/rtl8188eus/issues/91)

As commented in #91 . 
I can now compile it, but I am still having the same issue.
Any idea?
Would you need additional information?

escelikset, 2021-05-12:
ı am still having the same problem with you on Raspberry Pi 3B+ and couldn't figure it out

Miha22, 2021-05-21:
whats your chip number on wifi adapter? You can find yor chip by quering FCC ID of the device on: [Check chip](https://fccid.io/) and then go to **internal photos** and take a look which chip model is written on top of chip.

escelikset, 2021-05-21:
> whats your chip number on wifi adapter? You can find your chip by quering FCC ID of the device on: [Check chip](https://fccid.io/) and then go to **internal photos** and take a look which chip model is written on top of chip.

https://fccid.io/TE7WN722NV2

- REALTEK RTL8188EUS