aircrack-ng/rtl8188eus #90: error when compile android kernel

homeauto, 2020-10-05:
../drivers/staging/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1116:6: error: too many arguments to function call, expected 3, have 8
                        , pmlmepriv->assoc_req + sizeof(struct rtw_ieee80211_hdr_3addr) + 2
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/net/cfg80211.h:5662:6: note: 'cfg80211_roamed' declared here
void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
     ^
  CC [M]  drivers/staging/rtl8188eus/hal/hal_dm_acs.o
  CC [M]  drivers/staging/rtl8188eus/hal/hal_btcoex_wifionly.o
1 error generated.
../scripts/Makefile.build:361: recipe for target 'drivers/staging/rtl8188eus/os_dep/linux/ioctl_cfg80211.o' failed
make[3]: *** [drivers/staging/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Inväntar oavslutade jobb...
../scripts/Makefile.build:676: recipe for target 'drivers/staging/rtl8188eus' failed
make[2]: *** [drivers/staging/rtl8188eus] Error 2
make[2]: *** Inväntar oavslutade jobb...

been trying diffrent toolchains and got this error?

drygdryg, 2020-10-15:
What is the kernel version? I noticed problems with the build on older kernels (3.18.x)

homeauto, 2020-10-20:
I'm running 4.14 version

cyberknight777, 2021-06-03:
@homeauto did your issue get fixed ? i'm getting this same issue with kernel 4.9

niamotullah, 2021-06-03:
@homeauto @cyberknight777 Getting same issue on rtl8188eus and rtl8812au with 4.9 kernel

cyberknight777, 2021-06-18:
i got the issue resolved

Time1apse, 2022-04-14:
> i got the issue resolved

tell how pls

Time1apse, 2022-04-14:
found solution from this forum https://forums.kali.org/showthread.php?37911-Getting-RTL8188-to-work-with-Kali-in-monitor-mode with just deleting some code bcs i'm compiling with rtl8188 drivers