aircrack-ng/rtl8188eus #91: Error during make

TheBallOfAeolus, 2020-10-11:
Errors while compiling.
I have tried both v5.3.9 and master

Branch v5.3.9
```
username@kali:~/dev$ git clone -b v5.3.9 https://github.com/aircrack-ng/rtl8188eus.git
Cloning into 'rtl8188eus'...
remote: Enumerating objects: 85, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 2076 (delta 31), reused 33 (delta 10), pack-reused 1991
Receiving objects: 100% (2076/2076), 5.52 MiB | 6.37 MiB/s, done.
Resolving deltas: 100% (966/966), done.

username@kali:~/dev$ cd rtl8188eus/

username@kali:~/dev/rtl8188eus$ make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v8l+/build M=/home/username/dev/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v8l+'
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_security.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_io.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ap.o
/home/username/dev/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/home/username/dev/rtl8188eus/core/rtw_ap.c:195:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  195 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rson.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_odm.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rm.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /home/username/dev/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/os_intfs.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/usb_intf.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/ioctl_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/xmit_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/mlme_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/recv_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
/home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9805:2: error: #endif without #if
 9805 | #endif /* CONFIG_IOCTL_CFG80211 */
      |  ^~~~~
make[2]: *** [scripts/Makefile.build:304: /home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1522: _module_/home/username/dev/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.118-Re4son-v8l+'
make: *** [Makefile:2059: modules] Error 2
```

master
```
username@kali:~/dev/rtl8188eus$ cd ..

username@kali:~/dev$ rm -Rf rtl8188eus/

username@kali:~/dev$ git clone https://github.com/aircrack-ng/rtl8188eus.git
Cloning into 'rtl8188eus'...
remote: Enumerating objects: 85, done.
remote: Counting objects: 100% (85/85), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 2076 (delta 31), reused 33 (delta 10), pack-reused 1991
Receiving objects: 100% (2076/2076), 5.52 MiB | 5.49 MiB/s, done.
Resolving deltas: 100% (966/966), done.

username@kali:~/dev$ cd rtl8188eus/

username@kali:~/dev/rtl8188eus$ make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v8l+/build M=/home/username/dev/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v8l+'
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_security.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_io.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ap.o
/home/username/dev/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/home/username/dev/rtl8188eus/core/rtw_ap.c:195:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  195 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rson.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_odm.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rm.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /home/username/dev/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/os_intfs.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/usb_intf.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/ioctl_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/xmit_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/mlme_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/recv_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
/home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:9805:2: error: #endif without #if
 9805 | #endif /* CONFIG_IOCTL_CFG80211 */
      |  ^~~~~
make[2]: *** [scripts/Makefile.build:304: /home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1522: _module_/home/username/dev/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.118-Re4son-v8l+'
make: *** [Makefile:2059: modules] Error 2
username@kali:~/dev/rtl8188eus$
``` 

Could you help me out please?
Should I do something different?
Regards

kimocoder, 2020-10-11:
git pull && make -j4 && make install

TheBallOfAeolus, 2020-10-12:
Thank you @kimocoder 
because I did not remember which branch I have been using I decided to start from scratch.
I can compile, install, but still the same issue as per previous bug #86 

Here the results:
```
username@kali:~/dev$ rm -Rf rtl8188eus/

username@kali:~/dev$ git clone -b v5.3.9 https://github.com/aircrack-ng/rtl8188eus.git
Cloning into 'rtl8188eus'...
remote: Enumerating objects: 90, done.
remote: Counting objects: 100% (90/90), done.
remote: Compressing objects: 100% (76/76), done.
remote: Total 2081 (delta 34), reused 39 (delta 13), pack-reused 1991
Receiving objects: 100% (2081/2081), 5.52 MiB | 971.00 KiB/s, done.
Resolving deltas: 100% (969/969), done.

username@kali:~/dev$ cd rtl8188eus/

username@kali:~/dev/rtl8188eus$ make -j4
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v8l+/build M=/home/username/dev/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v8l+'
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_security.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_io.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_ap.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/username/dev/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rson.o
/home/username/dev/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/home/username/dev/rtl8188eus/core/rtw_ap.c:195:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  195 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_odm.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rm.o
  CC [M]  /home/username/dev/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /home/username/dev/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/os_intfs.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/usb_intf.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/ioctl_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/xmit_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/mlme_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/recv_linux.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/username/dev/rtl8188eus/os_dep/linux/rtw_cfgvendor.o
[... 134 lines cut ...]
  *-network:2
       description: Ethernet interface
       physical id: 3
       logical name: eth0
       serial: dc:a6:32:75:81:4c
       size: 1Gbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=bcmgenet driverversion=v2.0 duplex=full ip=192.168.10.144 link=yes multicast=yes port=MII speed=1Gbit/s

username@kali:~$ sudo ifconfig wlan0 down

username@kali:~$ sudo airmon-ng check kill
Killing these processes:

    PID Name
    384 dhclient
    457 wpa_supplicant

username@kali:~$ sudo iw dev wlan0 set type managed
command failed: No such device (-19)
```

I guess we can close this ticket because we can now compile it!
Thank you again!

viraj45, 2021-07-05:
I have same problem . Kali 2021.2  v5.10.46

closed 2022-03-19 by kimocoder