aircrack-ng/rtl8188eus #95: TP-LINK TL-WN722N V2.1 Monitor Mode Problem in Kali Linux

beesuns, 2020-10-30:
### I am running Kali Linux on a Dell laptop.
    kali@kali:~$ sudo uname -a
    Linux kali 5.8.0-kali3-amd64 #1 SMP Debian 5.8.14-1kali1 (2020-10-13) x86_64 GNU/Linux
### My wifi adaptor is TP-LINK TL-WN722N V2.1.
    kali@kali:~$ sudo airmon-ng

    PHY     Interface       Driver          Chipset

    phy0    wlan0           ath10k_pci      Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
    null    wlan1           r8188eu         TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]

### I tried 3 ways to put the adaptor into monitor mode, unfortunately all failed.
### 1st try:
    kali@kali:~$ sudo airmon-ng check

    Found 2 processes that could cause trouble.
    Kill them using 'airmon-ng check kill' before putting
    the card in monitor mode, they will interfere by changing channels
    and sometimes putting the interface back in managed mode

    PID Name
    610 NetworkManager
    687 wpa_supplicant

    kali@kali:~$ sudo airmon-ng check kill

    Killing these processes:

    PID Name
    687 wpa_supplicant

    kali@kali:~$ sudo airmon-ng start wlan1

    PHY     Interface       Driver          Chipset

    phy0    wlan0           ath10k_pci      Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
    null    wlan1           r8188eu         TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]

    kali@kali:~$ sudo airmon-ng

    PHY     Interface       Driver          Chipset

    phy0    wlan0           ath10k_pci      Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
    null    wlan1           r8188eu         TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]
### 2nd try:
    kali@kali:~$ sudo ifconfig wlan1 down
    kali@kali:~$ sudo iwconfig wlan1 mode Monitor
    Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan1 ; Invalid argument.
    kali@kali:~$ sudo ifconfig wlan1 up
    kali@kali:~$ sudo iwconfig
    lo        no wireless extensions.

    eth0      no wireless extensions.

    wlan0     IEEE 802.11  ESSID:"SSS" 
          Mode:Managed  Frequency:5.26 GHz  Access Point: 74:85:C4:20:7B:20   
          Bit Rate=6 Mb/s   Tx-Power=30 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=46/70  Signal level=-64 dBm 
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:3   Missed beacon:0

    wlan1     IEEE 802.11bgn  ESSID:"SSS"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: 74:85:C4:20:7B:30   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0 
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=69/100  Signal level=67/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
### 3rd try:
    kali@kali:~$ sudo iw dev
    phy#0
        Unnamed/non-netdev interface
                wdev 0x3
                addr a4:fc:77:6c:6e:75
                type P2P-device
                txpower 0.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr a4:fc:77:6c:6e:75
                ssid SSS
                type managed
                channel 52 (5260 MHz), width: 80 MHz, center1: 5290 MHz
                txpower 30.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
    kali@kali:~$ sudo iwconfig
    lo        no wireless extensions.

    eth0      no wireless extensions.

    wlan0     IEEE 802.11  ESSID:"SSS" 
          Mode:Managed  Frequency:5.26 GHz  Access Point: 74:85:C4:20:7B:20   
          Bit Rate=6 Mb/s   Tx-Power=30 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=47/70  Signal level=-63 dBm 
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:3   Missed beacon:0

    wlan1     IEEE 802.11bgn  ESSID:"SSS"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: 74:85:C4:20:7B:30   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0 
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=70/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

    kali@kali:~$ sudo airmon-ng

    PHY     Interface       Driver          Chipset

    phy0    wlan0           ath10k_pci      Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)
    null    wlan1           r8188eu         TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]

    kali@kali:~$ sudo iw dev
    phy#0
        Unnamed/non-netdev interface
                wdev 0x3
                addr a4:fc:77:6c:6e:75
                type P2P-device
                txpower 0.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr a4:fc:77:6c:6e:75
                ssid SSS
                type managed
                channel 52 (5260 MHz), width: 80 MHz, center1: 5290 MHz
                txpower 30.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
    kali@kali:~$ sudo ip link set wlan1 down
    kali@kali:~$ sudo iw wlan1 set monitor none
    command failed: No such device (-19)
### Please help, Thanks!

drygdryg, 2020-10-30:
Seems that you using r8188eu in-kernel driver, which doesn't support nl80211 and monitor mode. Please read the installation instructions carefully and make sure that you have blacklisted the r8188eu driver.

beesuns, 2020-10-30:
> Seems that you using r8188eu in-kernel driver, which doesn't support nl80211 and monitor mode. Please read the installation instructions carefully and make sure that you have blacklisted the r8188eu driver.

Thank you for replying! Are you meaning the installation instruction for the wireless card? 
I did not install any drive, I just plugged in the USB card and it worked, despite the fact that it can not be put into monitor mode.

drygdryg, 2020-10-30:
> Are you meaning the installation instruction for the wireless card?

I meaning this driver README: https://github.com/aircrack-ng/rtl8188eus/blob/v5.3.9/README.md
Firstly you need to build the driver, you can do this with DKMS:
```
sudo ./dkms-install.sh
```
Then just create a file to blacklist in-kernel r8188eu driver:
```
sudo sh -c 'echo "blacklist r8188eu" >> /etc/modprobe.d/realtek.conf'
```
And after these actions, just restart your system.

beesuns, 2020-10-31:
@drygdryg I have problems when I tried out.
I first followed the aircrack-ng/rtl8188eus Howto build/install instructions:
```
root@kali:/home/kali/rtl8188eus# ls
BUILD_FOR_NETHUNTER.md  core  dkms.conf  dkms-install.sh  dkms-remove.sh  hal  include  Kconfig  Makefile  os_dep  platform  README.md  ReleaseNotes.pdf
root@kali:/home/kali/rtl8188eus# echo "blacklist r8188eu" >> "/etc/modprobe.d/realtek.conf"
root@kali:/home/kali/rtl8188eus# make && make install
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-kali3-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: *** /lib/modules/5.8.0-kali3-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2059: modules] Error 2
```
Then I tried your instructions:
```
root@kali:/home/kali/rtl8188eus# sudo ./dkms-install.sh
About to run dkms install steps...
./dkms-install.sh: line 16: dkms: command not found
./dkms-install.sh: line 17: dkms: command not found
./dkms-install.sh: line 18: dkms: command not found
Finished running dkms install steps.
root@kali:/home/kali/rtl8188eus# ls
BUILD_FOR_NETHUNTER.md  core  dkms.conf  dkms-install.sh  dkms-remove.sh  hal  include  Kconfig  Makefile  os_dep  platform  README.md  ReleaseNotes.pdf
root@kali:/home/kali/rtl8188eus# cat dkms-install.sh 
#!/bin/bash

if [[ $EUID -ne 0 ]]; then
  echo "You must run this with superuser priviliges.  Try \"sudo ./dkms-install.sh\"" 2>&1
  exit 1
else
  echo "About to run dkms install steps..."
fi

DRV_DIR=rtl8188eus
DRV_NAME=8188eu
DRV_VERSION=5.3.9

cp -r ../${DRV_DIR} /usr/src/${DRV_NAME}-${DRV_VERSION}

dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
RESULT=$?

echo "Finished running dkms install steps."

exit $RESULT
```
And I am not sure what to do next. Further more, if I want to try this on a Kali Linux guest running in VirtualBox, what should I do? Since once I reboot the system, the change I made will be erased? Right?

drygdryg, 2020-10-31:
You need to install requirements firstly:
`sudo apt install linux-headers-$(uname -r) dkms bc`

beesuns, 2020-11-01:
@drygdryg You are a lifesaver. It worked finally. 
I first tried
```
root@kali:/home/kali/Downloads/rtl8188eus# apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-headers-5.8.0-kali3-amd64 is already the newest version (5.8.14-1kali1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Then I tried
```
root@kali:/home/kali/Downloads/rtl8188eus# apt-get install linux-headers-$(uname -r) dkms bc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-compiler-gcc-10-x86 linux-headers-5.8.0-kali3-common linux-headers-amd64 linux-kbuild-5.8
Suggested packages:
  menu
The following NEW packages will be installed:
  bc dkms linux-compiler-gcc-10-x86 linux-headers-5.8.0-kali3-amd64 linux-headers-5.8.0-kali3-common linux-headers-amd64 linux-kbuild-5.8
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 12.9 MB of archives.
After this operation, 62.3 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 dkms all 2.8.3-4 [77.3 kB]
Get:2 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 bc amd64 1.07.1-2+b2 [109 kB]
Get:3 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-compiler-gcc-10-x86 amd64 5.8.14-1kali1 [902 kB]
Get:4 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-headers-5.8.0-kali3-common all 5.8.14-1kali1 [9,306 kB]                                       
Get:5 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-kbuild-5.8 amd64 5.8.14-1kali1 [1,145 kB]                                                     
Get:6 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-headers-5.8.0-kali3-amd64 amd64 5.8.14-1kali1 [1,407 kB]                                      
Get:7 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-headers-amd64 amd64 5.8.14-1kali1 [1,148 B]                                                   
Fetched 12.9 MB in 9s (1,402 kB/s)                                                                                                                                   
Selecting previously unselected package dkms.
(Reading database ... 297377 files and directories currently installed.)
Preparing to unpack .../0-dkms_2.8.3-4_all.deb ...
Unpacking dkms (2.8.3-4) ...
Selecting previously unselected package bc.
Preparing to unpack .../1-bc_1.07.1-2+b2_amd64.deb ...
Unpacking bc (1.07.1-2+b2) ...
Selecting previously unselected package linux-compiler-gcc-10-x86.
Preparing to unpack .../2-linux-compiler-gcc-10-x86_5.8.14-1kali1_amd64.deb ...
Unpacking linux-compiler-gcc-10-x86 (5.8.14-1kali1) ...
Selecting previously unselected package linux-headers-5.8.0-kali3-common.
Preparing to unpack .../3-linux-headers-5.8.0-kali3-common_5.8.14-1kali1_all.deb ...
Unpacking linux-headers-5.8.0-kali3-common (5.8.14-1kali1) ...
Selecting previously unselected package linux-kbuild-5.8.
Preparing to unpack .../4-linux-kbuild-5.8_5.8.14-1kali1_amd64.deb ...
Unpacking linux-kbuild-5.8 (5.8.14-1kali1) ...
Selecting previously unselected package linux-headers-5.8.0-kali3-amd64.
Preparing to unpack .../5-linux-headers-5.8.0-kali3-amd64_5.8.14-1kali1_amd64.deb ...
Unpacking linux-headers-5.8.0-kali3-amd64 (5.8.14-1kali1) ...
Selecting previously unselected package linux-headers-amd64.
Preparing to unpack .../6-linux-headers-amd64_5.8.14-1kali1_amd64.deb ...
Unpacking linux-headers-amd64 (5.8.14-1kali1) ...
Setting up linux-compiler-gcc-10-x86 (5.8.14-1kali1) ...
Setting up linux-kbuild-5.8 (5.8.14-1kali1) ...
Setting up bc (1.07.1-2+b2) ...
Setting up dkms (2.8.3-4) ...
Setting up linux-headers-5.8.0-kali3-common (5.8.14-1kali1) ...
Setting up linux-headers-5.8.0-kali3-amd64 (5.8.14-1kali1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 5.8.0-kali3-amd64:.
Setting up linux-headers-amd64 (5.8.14-1kali1) ...
Processing triggers for kali-menu (2020.3.2) ...
Processing triggers for man-db (2.9.3-2) ...
```
Then
```
[... 123 lines cut ...]
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/phydm_api.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/phydm_pow_train.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_mac.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_bb.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_rf.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/rtl8188e/halrf_8188e_ce.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/phydm_regconfig8188e.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/hal8188erateadaptive.o
  CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/phydm_rtl8188e.o
  CC [M]  /home/kali/Downloads/rtl8188eus/platform/platform_ops.o
  CC [M]  /home/kali/Downloads/rtl8188eus/core/rtw_mp.o
  LD [M]  /home/kali/Downloads/rtl8188eus/8188eu.o
  MODPOST /home/kali/Downloads/rtl8188eus/Module.symvers
  CC [M]  /home/kali/Downloads/rtl8188eus/8188eu.mod.o
  LD [M]  /home/kali/Downloads/rtl8188eus/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-kali3-amd64'
install -p -m 644 8188eu.ko  /lib/modules/5.8.0-kali3-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.8.0-kali3-amd64
```
I put it into monitor mode and it seems worked just fine.

fluffy-cakes, 2020-12-05:
> I meaning this driver README: https://github.com/aircrack-ng/rtl8188eus/blob/v5.3.9/README.md
> Firstly you need to build the driver, you can do this with DKMS:
>
> ```
> sudo ./dkms-install.sh
> ```
>
> Then [...]

Just an FYI;

When using Kali linux for Raspberry Pi using 2020.3, I never needed to perform the `sudo ./dkms-install.sh` step. However, since updating/reimaging to 2020.4 I found that I had to, and it fixed my problem similar to the OP's.

beesuns, 2020-12-30:
> When using Kali linux for Raspberry Pi using 2020.3, I never needed to perform the `sudo ./dkms-install.sh` step. However, since updating/reimaging to 2020.4 I found that I had to, and it fixed my problem similar to the OP's.

Any advice on how to use this adaptor on Linux raspberrypi 5.4.79-v71+ armv71 GUN/Linux?

fluffy-cakes, 2021-03-11:
> > When using Kali linux for Raspberry Pi using 2020.3, I never needed to perform the `sudo ./dkms-install.sh` step. However, since updating/reimaging to 2020.4 I found that I had to, and it fixed my problem similar to the OP's.
> 
> Any advice on how to use this adaptor on Raspberry Pi OS : `Linux raspberrypi 5.4.79-v71+ armv71 GUN/Linux`?
> And I also want to try it on kali linux arm image : `Linux kali 4.19.118-Re4son-v81+ aarch 64 GUN/Linux`.

sorry, belated reply. i found Pi OS as unsuitable for this type of use and resorted back to using Kali

Mugunthan656, 2022-01-10:
Make command not working.... 
I have updated and upgraded everything 
[image]

prometey1337, 2022-01-10:
> @drygdryg You are a lifesaver. It worked finally. I first tried
> 
> ```
> root@kali:/home/kali/Downloads/rtl8188eus# apt-get install linux-headers-$(uname -r)
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> linux-headers-5.8.0-kali3-amd64 is already the newest version (5.8.14-1kali1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> ```
> 
> Then I tried
> 
> ```
> root@kali:/home/kali/Downloads/rtl8188eus# apt-get install linux-headers-$(uname -r) dkms bc
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> The following additional packages will be installed:
>   linux-compiler-gcc-10-x86 linux-headers-5.8.0-kali3-common linux-headers-amd64 linux-kbuild-5.8
> Suggested packages:
>   menu
> The following NEW packages will be installed:
>   bc dkms linux-compiler-gcc-10-x86 linux-headers-5.8.0-kali3-amd64 linux-headers-5.8.0-kali3-common linux-headers-amd64 linux-kbuild-5.8
> 0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
> Need to get 12.9 MB of archives.
> After this operation, 62.3 MB of additional disk space will be used.
> Do you want to continue? [Y/n] y
> Get:1 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 dkms all 2.8.3-4 [77.3 kB]
> Get:2 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 bc amd64 1.07.1-2+b2 [109 kB]
> Get:3 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-compiler-gcc-10-x86 amd64 5.8.14-1kali1 [902 kB]
> Get:4 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-headers-5.8.0-kali3-common all 5.8.14-1kali1 [9,306 kB]                                       
> Get:5 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-kbuild-5.8 amd64 5.8.14-1kali1 [1,145 kB]                                                     
> Get:6 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-headers-5.8.0-kali3-amd64 amd64 5.8.14-1kali1 [1,407 kB]                                      
> Get:7 http://mirrors.neusoft.edu.cn/kali kali-rolling/main amd64 linux-headers-amd64 amd64 5.8.14-1kali1 [1,148 B]                                                   
> Fetched 12.9 MB in 9s (1,402 kB/s)                                                                                                                                   
> Selecting previously unselected package dkms.
> (Reading database ... 297377 files and directories currently installed.)
> Preparing to unpack .../0-dkms_2.8.3-4_all.deb ...
> Unpacking dkms (2.8.3-4) ...
> Selecting previously unselected package bc.
> Preparing to unpack .../1-bc_1.07.1-2+b2_amd64.deb ...
> Unpacking bc (1.07.1-2+b2) ...
> Selecting previously unselected package linux-compiler-gcc-10-x86.
> Preparing to unpack .../2-linux-compiler-gcc-10-x86_5.8.14-1kali1_amd64.deb ...
> Unpacking linux-compiler-gcc-10-x86 (5.8.14-1kali1) ...
> Selecting previously unselected package linux-headers-5.8.0-kali3-common.
> Preparing to unpack .../3-linux-headers-5.8.0-kali3-common_5.8.14-1kali1_all.deb ...
> Unpacking linux-headers-5.8.0-kali3-common (5.8.14-1kali1) ...
> Selecting previously unselected package linux-kbuild-5.8.
> Preparing to unpack .../4-linux-kbuild-5.8_5.8.14-1kali1_amd64.deb ...
> Unpacking linux-kbuild-5.8 (5.8.14-1kali1) ...
> Selecting previously unselected package linux-headers-5.8.0-kali3-amd64.
> Preparing to unpack .../5-linux-headers-5.8.0-kali3-amd64_5.8.14-1kali1_amd64.deb ...
> Unpacking linux-headers-5.8.0-kali3-amd64 (5.8.14-1kali1) ...
> Selecting previously unselected package linux-headers-amd64.
> Preparing to unpack .../6-linux-headers-amd64_5.8.14-1kali1_amd64.deb ...
> Unpacking linux-headers-amd64 (5.8.14-1kali1) ...
> Setting up linux-compiler-gcc-10-x86 (5.8.14-1kali1) ...
> Setting up linux-kbuild-5.8 (5.8.14-1kali1) ...
> Setting up bc (1.07.1-2+b2) ...
> Setting up dkms (2.8.3-4) ...
> Setting up linux-headers-5.8.0-kali3-common (5.8.14-1kali1) ...
> Setting up linux-headers-5.8.0-kali3-amd64 (5.8.14-1kali1) ...
> /etc/kernel/header_postinst.d/dkms:
> dkms: running auto installation service for kernel 5.8.0-kali3-amd64:.
> Setting up linux-headers-amd64 (5.8.14-1kali1) ...
> Processing triggers for kali-menu (2020.3.2) ...
> Processing triggers for man-db (2.9.3-2) ...
> ```
[... 130 lines cut ...]
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halphyrf_ce.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf_powertracking_ce.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf_powertracking.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/halrf_kfree.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_mac.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_bb.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_rf.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/halrf/rtl8188e/halrf_8188e_ce.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/phydm_regconfig8188e.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/hal8188erateadaptive.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/hal/phydm/rtl8188e/phydm_rtl8188e.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/platform/platform_ops.o
>   CC [M]  /home/kali/Downloads/rtl8188eus/core/rtw_mp.o
>   LD [M]  /home/kali/Downloads/rtl8188eus/8188eu.o
>   MODPOST /home/kali/Downloads/rtl8188eus/Module.symvers
[... cut ...]
>   LD [M]  /home/kali/Downloads/rtl8188eus/8188eu.ko
> make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-kali3-amd64'
> install -p -m 644 8188eu.ko  /lib/modules/5.8.0-kali3-amd64/kernel/drivers/net/wireless/
> /sbin/depmod -a 5.8.0-kali3-amd64
> ```
> 
> After a reboot, I put it into monitor mode and it seemed work just fine. Any ideas on how to do this on a Linux guest running in VirtualBox? Thanks!

Hi man, can u help me pls. When i start command make in rt8188eus directory, i have Error 2.

ValenJes, 2022-07-27:
Hello good day from Argentina. my name is Santiago. recently I was trying to install as you did, but from the "make" command I get an error as you can see in the attached image. if you could help me I would really appreciate it.