aircrack-ng/rtl8188eus #106: Make error  on rtl8188eus file

junnu-junaid, 2021-01-11:
During make make time im getting error as
/Lib/modules/5.9.0-kali-amd64/build: no such file or directory . Stop
Make file : 2059 : modules error 2 
... 
Even though i have installed all build essential generic but still im facing the same errors.

SmillerMP, 2021-01-12:
You have to install linux headers, look and search others issues about make error 2

junnu-junaid, 2021-01-12:
I have installed but still it says the same

MrEngman, 2021-01-13:
The build error is because a symbolic link to you kernel source is missing.

You need to add the symbolic link using command like below to point to where your kernel source directory is.

sudo ln -s /home/kernel/source  /lib/modules/$(uname -r)/build

closed 2022-03-19 by kimocoder