aircrack-ng/rtl8188eus #107: Not working in Fedora 33, kernel 5.10.7

mayconcancini, 2021-01-17:
After upgrade from kernel 5.10.6 to 5.10.7 in Fedora 33, and again make && sudo make install, this driver stopped working.

rfm83, 2021-01-18:
Same issue for rtl8821ce wireless driver.

mayconcancini, 2021-01-21:
After upgrading to the 5.10.8 kernel, the driver returned to normal operation.

closed 2021-01-21 by mayconcancini

galda01, 2021-02-04:
Same issue. 

Kernel: 5.10.11-200.fc33.x86_64

It was working prior to the upgrade from Fedora 32 to Fedora 33. 

Reproduce:
# ifconfig wlp0s20f3 down
# airodump-ng wlp0s20f3
# <can't see any wifi traffic>

Also, at this point, the system can't get WIFI up again. I'm sure there's a way but i haven't figure it out yet. Needs a reboot until i do figure it out.