aircrack-ng/rtl8188eus #111: https://github.com/aircrack-ng/rtl8188eus

Mojtaba81425, 2021-02-16:
/root/rtl8188eus-5.7.6.1/rtl8188eus/os_dep/osdep_service.c: In function ‘storeToFile’:
/root/rtl8188eus-5.7.6.1/rtl8188eus/os_dep/osdep_service.c:2293:4: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2293 |    set_fs(get_ds());
      |    ^~~~~~
      |    sget_fc
/root/rtl8188eus-5.7.6.1/rtl8188eus/os_dep/osdep_service.c:2293:11: error: implicit declaration of function ‘get_ds’; did you mean ‘get_da’? [-Werror=implicit-function-declaration]
 2293 |    set_fs(get_ds());
      |           ^~~~~~
      |           get_da
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.build:284: /root/rtl8188eus-5.7.6.1/rtl8188eus/os_dep/osdep_service.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:1817: /root/rtl8188eus-5.7.6.1/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-kali3-amd64'
make: *** [Makefile:2313: modules] Error 2

hello guys can you help me with this error thanks.

ibnuvboys, 2021-03-08:
I'm the same as that

MrRob0-X, 2021-06-06:
@Mojtaba81425 you have the solution, actually u can see what compiler tell you

like here ` set_fs’; did you mean ‘sget_fc’? `
also `‘get_ds’; did you mean ‘get_da’?`

frank8922, 2021-10-18:
> @Mojtaba81425 you have the solution, actually you can see what compiler tell you
> 
> like here `set_fs’; did you mean ‘sget_fc’?` also `‘get_ds’; did you mean ‘get_da’?`

how would you do that?

frank8922, 2021-10-18:
|
/home/ubuntu/rtl8822bu/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/ubuntu/rtl8822bu/os_dep/osdep_service.c:2020:10: error: too few arguments to function ‘get_da’
 2020 |   set_fs(get_da());
      |          ^~~~~~
 
This is what I get now. I have no idea what I am doing. I just want to wifi work

frank8922, 2021-10-18:
/home/ubuntu/rtl8822bu/os_dep/osdep_service.c:2050:4: error: too few arguments to function ‘sget_fc’
 2050 |    sget_fc(get_ds());
      |    ^~~~~~~
In file included from ./include/linux/compat.h:17,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:37,
                 from /home/ubuntu/rtl8822bu/include/osdep_service_linux.h:35,
                 from /home/ubuntu/rtl8822bu/include/osdep_service.h:41,
                 from /home/ubuntu/rtl8822bu/include/drv_types.h:32,
                 from /home/ubuntu/rtl8822bu/os_dep/osdep_service.c:24:
./include/linux/fs.h:2300:21: note: declared here
 2300 | struct super_block *sget_fc(struct fs_context *fc,
      |                     ^~~~~~~

It worked for a bit, then crashed with this

frank8922, 2021-10-18:
https://github.com/jeremyb31/rtl8822bu this the one I got. I tried following the instructions and got the same error. I googled the error and saw this post with the exact same error.

Zyad-Elsayed, 2021-12-04:
hello, can you help me with this 
when I enter this command 
git clone https://github.com/aircrack-ng/rtl8188eus
I get 
Cloning into 'rtl8188eus'...
fatal: unable to access 'https://github.com/aircrack-ng/rtl8188eus/': Could not resolve host: github.com
so what should I do

MrRob0-X, 2021-12-04:
> hello, can you help me with this when I enter this command git clone https://github.com/aircrack-ng/rtl8188eus I get Cloning into 'rtl8188eus'... fatal: unable to access 'https://github.com/aircrack-ng/rtl8188eus/': Could not resolve host: github.com so what should I do

[image]

working as good, make sure to check your internet.

BlackCorp1, 2022-06-19:
alguem pode me ajudar por favor a resolver esses erros IRRITANTES do comando "make"??? sou novato no linux!!! quais comandos devo usar??? obrigado...

(root Black)-[-/rt18188eus] L# make && make install make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-kali3-amd64/build M=/root/r t18188eus modules make[1]: Entrando no diretório '/usr/src/linux-headers-5.17.0-kali3-amd64',
CC [M] /root/rt18188eus/os_dep/osdep_service.o /root/rt18188eus/os_dep/osdep service.c: In function 'thread_exit': /root/rt18188eus/os_dep/osdep_service.c:1271:9: error: implicit declaration of fun ction 'complete_and_exit' [-Werror=implicit-function-declaration] 1271 |
complete_and_exit(comp, 0);
/root/rt18188eus/os_dep/osdep_service.c: In function 'rtw_change_ifname': /root/rt18188eus/os_dep/osdep_service.c:2450:28: warning: passing argument 1 of '. rtw_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-quali, fiers] 2450 |
_rtw_memcpy(pnetdev-dev_addr, adapter_mac_addr(padapter), ETH_ALE
N);
/root/rt18188eus/os_dep/osdep_service.c:899:24: note: expected 'void *' but argume int is of type 'const unsigned char *
899 | void _rtw_memcpy(void *dst, const void *src, u32 sz),
cc1: some warnings being treated as errors make[2]: *** [/usr/src/linux-headers-5.17.0-kali3-common/scripts/Makefile.build:29 3: /root/rt18188eus/os_dep/osdep_service.o] Erro 1 make[1]: *** [/usr/src/linux-headers-5.17.0-kali3-common/Makefile:1855: /root/rt18 188eus] Erro 2 make[1]: Saindo do diretório '/usr/src/linux-headers-5.17.0-kali3-amd64' make: *** [Makefile:2058: modules] Erro 2