aircrack-ng/rtl8188eus #117: can't change inject rate?

didadidadida, 2021-03-26:
how to change inject rate? it is 1mbps although "pattrib->rate = MGN_24M;".and " iw dev wlan0 set bitrates mcs-2.4 1
command failed: Operation not supported (-95) ".

bakwc, 2021-08-25:
You can change tx rate by modifing appropriate field in radiotap header, when creating packet.
Driver will use this field to transmitt with appropriate rates.
Most hardware don't support `iw dev wlan0 set bitrates`.

```
static uint8_t u8aRadiotapHeader[] = {
        0x00, 0x00,             // <-- radiotap version
        0x0c, 0x00,             // <- radiotap header length
        0x04, 0x80, 0x00, 0x00, // <-- radiotap present flags (rate + tx flags)
        0x00,                   // datarate (will be overwritten later in packet_header_init)
        0x00,                   // ??
        0x00, 0x00              // ??
};
```

`u8aRadiotapHeader` - this is data rate. Eg, if you want a 48 mbit data rate, you should set

`u8aRadiotapHeader[8] = 0x60`

Vito-Swift, 2023-12-13:
Realtek drivers which have the source file `core/rtw_xmit.c` would call `rtw_monitor_xmit_entry()` to send an inject frame to the Wi-Fi device.

First, the program will neglect all radiotap payloads except the rtap_version and rtap_length fields in the header. With that being said, no matter what datarate is specified by the user, it will not affect the transmission process. 

Additionally, in Line 4006 of the file ([link](https://github.com/aircrack-ng/rtl8188eus/blob/3fae7237ba121f1169e9a2ea55040dc123697d3b/core/rtw_xmit.c#L4406)), the program will just use a *fixed* data rate `pattrib->rate = MGN_24M`. So there are two options one can do to achieve data rate selection in injection frames:
- Change the override datarate `pattrib->rate` to other literals like `MGN_54M`.
- Correct the radiotap header parsing and use the rate field in the header to select data rate. I maintained a patch of this kind in [here](https://github.com/Vito-Swift/rtl8814au-ext/commit/4a375393ed8fffc39b7c9497bc2a41342c808d5a), but the patch does not guarantee to be bugfree in general cases.