aircrack-ng/rtl8188eus #128: Can't Compile in Kernel 5.10

KevinMagnifico260, 2021-05-06:
[image]

akhilps96, 2021-05-13:
Hey try using make clean command first then use make to compile, I had the same issue this worked for me

KevinMagnifico260, 2021-05-20:
> Hey try using make clean command first then use make to compile, I had the same issue this worked for me

Thanks.

hadibuxm, 2021-07-24:
after running make clean, it still shows same issue

hadibuxm, 2021-07-24:
make cleaned worked but now dkms status is showing following error 
**`8188eu 5.3.v9 5.10.0-kali-amd64,+x86_64:+installed+(WARNING!+Diff+between+built+and+installed+module!)`**

shashankx86, 2022-03-04:
use this https://github.com/drygdryg/rtl8188eus

NickTheNeko2002, 2022-09-18:
> Hey try using make clean command first then use make to compile, I had the same issue this worked for me

i tried it and doesn't work with me (OS Linux Arch, Kernel 5.10.90-1-lts)
[image]

shashankx86, 2022-09-18:
> > Hey try using make clean command first then use make to compile, I had the same issue this worked for me
> 
> i tried it and doesn't work with me (OS Linux Arch, Kernel 5.10.90-1-lts) [image]

use this https://github.com/drygdryg/rtl8188eus

NickTheNeko2002, 2022-09-19:
> > > Hey try using make clean command first then use make to compile, I had the same issue this worked for me
> > 
> > 
> > i tried it and doesn't work with me (OS Linux Arch, Kernel 5.10.90-1-lts) [image]
> 
> use this https://github.com/drygdryg/rtl8188eus

i'm still having the same issue, i think it's gcc causing the issue but i have no idea how to fix it