aircrack-ng/rtl8188eus #129: error cut: /etc/redhat-release: No such file or directory

mprabhakarece, 2021-05-20:
Hi
I am getting above error while trying to install driver to my Jetson Xavier board with Ubuntu 18 OS
Kindly do needful

Miha22, 2021-05-21:
Its not a problem, it can work with out it. Make sure you have installed linux headers (kali7-amd64)

ENSANITYel, 2022-03-04:
how do i install kali7-amd64

ENSANITYel, 2022-03-04:
i have installed headers but 5.16.0 kali1-amd64

savetheword, 2022-03-12:
Hi bro please help me

savetheword, 2022-03-12:
> i have installed headers but 5.16.0 kali1-amd64

you installed ?

savetheword, 2022-03-12:
> how do i install kali7-amd64

use this link http://old.kali.org/kali/pool/main/l/linux/

savetheword, 2022-03-13:
> i have installed headers but 5.16.0 kali1-amd64

 any problems you facing https://github.com/drygdryg/rtl8188eus use this repository

savetheword, 2022-03-13:
> how do i install kali7-amd64

What issue you facing?