aircrack-ng/rtl8188eus #130: Won't enable monitoring on RTL8188EUS error: /etc/redhat-release: No such file or directory

eiger3970, 2021-05-23:
`┌──(root💀kali)-[/home/kali/rtl8188eus]
└─# uname -r
5.10.0-kali3-amd64`

`┌──(root💀kali)-[/home/kali/rtl8188eus]
└─# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID abcd:1234 LogiLink UDisk flash drive
Bus 001 Device 002: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

`──(root💀kali)-[/home/kali/rtl8188eus]
└─# iwconfig                                                                                                           127 ⨯
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.437 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth1      no wireless extensions.
`

`┌──(root💀kali)-[/home/kali/rtl8188eus]
└─# make && make install
cut: /etc/redhat-release: No such file or directory
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali3-amd64/build M=/home/kali/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-kali3-amd64'
cut: /etc/redhat-release: No such file or directory
/bin/sh: 1: bc: not found
  CC [M]  /home/kali/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_security.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_io.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_ap.o
  CC [M]  /home/kali/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/kali/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/kali/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_rson.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_btcoex.o
[... 82 lines cut ...]
  CC [M]  /home/kali/rtl8188eus/hal/phydm/halrf/halrf.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_mac.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_bb.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_rf.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/halrf/rtl8188e/halrf_8188e_ce.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/rtl8188e/phydm_regconfig8188e.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/rtl8188e/hal8188erateadaptive.o
  CC [M]  /home/kali/rtl8188eus/hal/phydm/rtl8188e/phydm_rtl8188e.o
  CC [M]  /home/kali/rtl8188eus/platform/platform_ops.o
  CC [M]  /home/kali/rtl8188eus/core/rtw_mp.o
  LD [M]  /home/kali/rtl8188eus/8188eu.o
cut: /etc/redhat-release: No such file or directory
/bin/sh: 1: bc: not found
  MODPOST /home/kali/rtl8188eus/Module.symvers
  CC [M]  /home/kali/rtl8188eus/8188eu.mod.o
  LD [M]  /home/kali/rtl8188eus/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-kali3-amd64'
cut: /etc/redhat-release: No such file or directory
/bin/sh: 1: bc: not found
install -p -m 644 8188eu.ko  /lib/modules/5.10.0-kali3-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.10.0-kali3-amd64`

alkock, 2021-06-17:
Same goes for me, running Kali Linux installed on a hardware device.

Mayank50082785, 2022-01-24:
same +1

error:-
┌──(root💀kali)-[~/Desktop/rtl8188eus]
└─# make && sudo make install
cut: /etc/redhat-release: No such file or directory
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-kali4-amd64/build M=/root/Desktop/rtl8188eus  modules
make[1]: *** /lib/modules/5.14.0-kali4-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2062: modules] Error 2

savetheword, 2022-03-12:
Same problem i have

> same +1
>
> error:- ┌──(root💀kali)-[~/Desktop/rtl8188eus] └─# make && sudo make install cut: /etc/redhat-release: No such file or directory /bin/sh: 1: bc: not found make ARCH=x86_64 CROSS_COMPILE= -C [...]