aircrack-ng/rtl8188eus #131: Make command issues

galloq, 2021-05-27:
Hello newbie right here I am currently trying to compile the latest version of this repo, and getting the next issues: 

[image]

FYI: Before trying to compile I followed these steps to ensure there were no issues with libraries(still got the above error), based on this youtube video: https://www.youtube.com/watch?v=tYnjMiTTdms

`sudo apt update: 
sudo apt upgrade -y: 
sudo apt install bc: 
Restart Kali Linux: 
sudo apt-get install build-essential: 
sudo apt-get install libelf-dev:
sudo apt-get install linux-headers-`uname -r`: 
sudo apt install dkms: 
sudo rmmod r8188eu.ko: 
git clone https://github.com/aircrack-ng/rtl818... 
cd rtl8188eus: 
sudo -i: 
echo "blacklist r8188eu" to  "/etc/modprobe.d/realtek.conf": 
exit: 
sudo reboot: 
sudo apt update: 
cd rtl8188eus: 
sudo make: (failed at this point)`

Thank you very much for your help!

galloq, 2021-05-27:
Updating this issue this was solved by compiling the branch OpenHD-v5.3.9 instead of v5.3.9.
I am still using the same versions of kali linux and same kernel version.

jesse-s07, 2021-07-01:
O vale para distribuição ubunto linux?

jesse-s07, 2021-07-01:
A mesma solução vale também  para distribuição ubunto linux?

galloq, 2021-07-12:
Não testei a solução lá, mas diga-nos se funcionou.

I haven't tested the solution there, but please let us know if it worked.
In case it does or it doesn't please let us know as well, try to include the ubuntu version as it's an important detail. 

Translated with google / Traduzido com google

jesse-s07, 2021-07-29:
eu testei aqui e funcionou muito bem,  não só no Ubunto 64 bits 20.04.2.0 LTS, mas também no sistema Bodhi Linux 64 bits, 5.4.0-72.

closed 2022-03-19 by kimocoder