aircrack-ng/rtl8188eus #132: Can not make on macbook air with m1 chip

Khumayun, 2021-06-01:
[image]
I have trier installing headers as explained in issue 115[https://github.com/aircrack-ng/rtl8188eus/issues/115](url).

While typing sudo apt install linux-headers-*** it pops up several choices: 

- `linux-headers-5.10.0-kali7-arm64`
- `linux-headers-5.10.0-kali7-rt-arm64 `
- `linux-headers-5.10.0-kali7-cloud-arm64`
- `linux-headers-arm64`
- `linux-headers-5.10.0-kali7-common`
- `linux-headers-cloud-arm64`
- `linux-headers-5.10.0-kali7-common-rt`
- `linux-headers-rt-arm64`
but no linux-headers-5.10.0-kali3-*** as it is required by make command.

Also there exists in /lib/modules/5.10.0-kali7-arm64/build  both `5.10.0-kali3-arm64/` and `5.10.0-kali7-arm64/`

I am using Parallels to run virtual machine.

SolArt7, 2022-03-23:
You can find a solution in this thread
https://github.com/aircrack-ng/rtl8188eus/issues/64

taziksh, 2024-08-12:
> You can find a solution in this thread #64

@SolArt7 that's not for Mac