aircrack-ng/rtl8188eus #136: makefile:2244:modules error

Arsalan692, 2021-07-25:
[image]

i am having this issue and I cannot resolve how can I fix it

voldmort21, 2021-08-07:
you need to install kernel headers

Jibun-no-Kage, 2022-04-27:
Load the kernel headers.  It is common issue.  You can determine which version of the headers to install via the current kernel.

$ uname -a
Linux eccetric 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux

Typically your distribution should have a package that includes the applicable headers.  For example on Debian 11...

$ dpkg -l | grep kernel-headers
ii  raspberrypi-kernel-headers           1:1.20220331-1                   armhf        Header files for the Raspberry Pi Linux kernel

The above example is for an ARM processor, but i386/x64 is the same process, just named for the kernel version and hardware applicable.

Jibun-no-Kage, 2022-04-27:
For reference to anyone that would like a step by step example for x86-64.  For ARM you will need to change the Makefile...

# lsusb | grep -i rtl
Bus 002 Device 005: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

(Login as root)
# cd /root
# git clone https://github.com/aircrack-ng/rtl8188eus.git
Cloning into 'rtl8188eus'...
remote: Enumerating objects: 2298, done.
remote: Counting objects: 100% (201/201), done.
remote: Compressing objects: 100% (111/111), done.
remote: Total 2298 (delta 96), reused 185 (delta 89), pack-reused 2097
Receiving objects: 100% (2298/2298), 5.87 MiB | 7.82 MiB/s, done.
Resolving deltas: 100% (1091/1091), done.

# apt update
# apt install dkms

You can get the version of the driver from the dkms.conf file or some times in the read-me file...
# cp  -rf ./rtl8188eus/ /usr/src/rtl8188eu-5.3.9
# dkms add -m rtl8188eu -v 5.3.9
Creating symlink /var/lib/dkms/rtl8188eu/5.3.9/source ->
                 /usr/src/rtl8188eu-5.3.9

DKMS: add completed.

# dkms build -m rtl8188eu -v 5.3.9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j5 KVER=5.10.0-13-amd64 KSRC=/lib/modules/5.10.0-13-amd64/build.............
cleaning build area...

DKMS: build completed.

# dkms install -m rtl8188eu -v 5.3.9
Module rtl8188eu/5.3.9 already built for kernel 5.10.0-13-amd64/4
root@crazy:~# dkms install -m rtl8188eu -v 5.3.9

8188eu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.0-13-amd64/updates/dkms/

depmod....

DKMS: install completed.

At this point check to see if the wireless adapter is connected to an access point, i.e. you wireless router.  Check if interface is blocked, sometimes some distributions do this...

# apt install rfkill
# rfkill

Look for interface of wireless adapter, if it is present, but does not have an ip address, continue...
# ifconfig

See if the wireless adapter can scan for available networks...  Assuming the wireless adapter starts with 'wlx'...  Versus 'wlan'...
# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enp5s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether XX:XX:XX:XX:XX:XX brd ff:ff:ff:ff:ff:ff
3: wlxXXXXXXXXXXXX: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DORMANT group default qlen 1000
    link/ether XX:XX:XX:XX:XX:XX:XX brd ff:ff:ff:ff:ff:ff

# apt install wireless-tools
# iw $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') info
command failed: No such device (-19)

Opps not alive yet... Try scan...
# iwlist $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') scan

If any cells or wireless networks found, your adapter is at least scanning and receiving available access points/networks.
Now, try to force adapter to connect to your specific wireless router, you will need your network name and network password, defined in your /etc/wpa_supplicant/wpa_supplicant.conf, this means your wpa_supplicant configuration will need to be present, for this step.  This example has the expectation that you have DHCP setup, i.e. your environment provides IP addresses on demand...

# wpa_supplicant -B -D wext -i $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') -c /etc/wpa_supplicant/wpa_supplicant.conf
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument

The output of the above command will vary based on your specific hardware and configuration.  But if it works, you next step will show the wireless adapter is now associated to your wireless router...

# iwconfig $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') | grep 'Access Point'
          Mode:Managed  Frequency=2.462 GHz  Access Point: Not-Associated
Versus...

If it shows the access point as not associated, your adapter is not yet talking to your wireless router, likely your wpa_supplication.conf file is wrong, i.e. your wireless network name and password maybe wrong.

Sometimes rebooting or unplugging and re-plugging your wireless (USB) adapter can wake up the driver and complete the association.

Sometimes you will need to make sure the firmware for the adapter is present on the system as well.

#